espressif/ESP8266_RTOS_SDK #930: Compilation Error in Hello World Example

kozmotronik, 2020-07-05:
## Environment

- Development Kit: ESP8266 AMICA Node MCU
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:      Make
- Operating System:     Ubuntu
- Power Supply:         USB

## Problem Description
I am trying to get started with ESP8266 AMICA Node MCU development kit using ESP8266 RTOS SDK for 4 days. I struggled a lot with settting up the required environment and getting it work. I went step by step as per instructions in esp official docs: https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/#get-started-get-esp-id. I got up to Configure topic after a lot of struggling with environment path etc. and solved all problems so far. But when I am trying to flash the Hello world example into the chip using make flash command, I am getting these errors:

/home/kozmotronik/esp/ESP8266_RTOS_SDK/components/esp8266/include/driver/soc.h: Assembler messages:
/home/kozmotronik/esp/ESP8266_RTOS_SDK/components/esp8266/include/driver/soc.h:103: Error: invalid register 'intclear' for 'wsr' instruction
make[1]: *** [/home/kozmotronik/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:292: source/task_wdt.o] Error 1
make: *** [/home/kozmotronik/esp/ESP8266_RTOS_SDK/make/project.mk:571: component-esp8266-build] Error 2

I couldn't figure out how to fix this problem. In addition I didn't change or modify any source file neither in the hello world example nor in the SDK source files, everything is as installed. So what is wrong?

### Expected Behavior
Compiles an flashes the node mcu esp8266 chip

### Actual Behavior
Compilation errors.

## Debug Logs
Here I copy and past the output of the "make flash" command from my command line:
```
kozmotronik@kozmotronik-GE60-0NC-0ND:~/esp/hello_world$ make flash
Toolchain path: /home/kozmotronik/esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: crosstool-ng-1.22.0-100-ge567ec7
Compiler version: 5.2.0
Python requirements from /home/kozmotronik/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/spi_flash/src/spi_flash.o
CC build/bootloader/spi_flash/src/spi_flash_raw.o
CC build/bootloader/spi_flash/port/port.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/esp8266/source/ets_printf.o
CC build/bootloader/esp8266/source/crc.o
AR build/bootloader/esp8266/libesp8266.a
LD build/bootloader/bootloader.elf
esptool.py v2.4.0
Building partitions from /home/kozmotronik/esp/ESP8266_RTOS_SDK/components/partition_table/partitions_singleapp.csv...
Project is not inside a git repository, or git repository has no commits
will not use 'git describe' to determine PROJECT_VER.
App "hello-world" version: 1
CC build/app_update/esp_ota_ops.o
CC build/app_update/esp_app_desc.o
AR build/app_update/libapp_update.a
CC build/bootloader_support/src/efuse.o
CC build/bootloader_support/src/bootloader_flash.o
CC build/bootloader_support/src/flash_qio_mode.o
CC build/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader_support/src/bootloader_init.o
CC build/bootloader_support/src/bootloader_common.o
CC build/bootloader_support/src/secure_boot.o
CC build/bootloader_support/src/bootloader_clock.o
CC build/bootloader_support/src/bootloader_random.o
CC build/bootloader_support/src/flash_partitions.o
CC build/bootloader_support/src/esp_image_format.o
CC build/bootloader_support/src/bootloader_sha.o
CC build/bootloader_support/src/flash_encrypt.o
CC build/bootloader_support/src/bootloader_utility.o
AR build/bootloader_support/libbootloader_support.a
CC build/coap/libcoap/src/address.o
CC build/coap/libcoap/src/async.o
CC build/coap/libcoap/src/block.o
CC build/coap/libcoap/src/coap_time.o
CC build/coap/libcoap/src/debug.o
CC build/coap/libcoap/src/encode.o
CC build/coap/libcoap/src/hashkey.o
CC build/coap/libcoap/src/mem.o
CC build/coap/libcoap/src/net.o
CC build/coap/libcoap/src/option.o
CC build/coap/libcoap/src/pdu.o
CC build/coap/libcoap/src/resource.o
CC build/coap/libcoap/src/str.o
CC build/coap/libcoap/src/subscribe.o
CC build/coap/libcoap/src/uri.o
CC build/coap/port/coap_io_socket.o
AR build/coap/libcoap.a
AR build/console/libconsole.a
CC build/esp-tls/esp_tls.o
CC build/esp-tls/esp_tls_mbedtls.o
AR build/esp-tls/libesp-tls.a
AR build/esp-wolfssl/libesp-wolfssl.a
CC build/esp8266/driver/hw_timer.o
CC build/esp8266/driver/ir_rx.o
CC build/esp8266/driver/ir_tx.o
CC build/esp8266/driver/uart.o
CC build/esp8266/driver/hspi_logic_layer.o
CC build/esp8266/driver/pwm.o
CC build/esp8266/driver/i2s.o
CC build/esp8266/driver/spi.o
CC build/esp8266/driver/ledc.o
CC build/esp8266/driver/gpio.o
CC build/esp8266/driver/i2c.o
CC build/esp8266/driver/adc.o
CC build/esp8266/source/reset_reason.o
CC build/esp8266/source/esp_fsleep.o
CC build/esp8266/source/task_wdt.o
/home/kozmotronik/esp/ESP8266_RTOS_SDK/components/esp8266/include/driver/soc.h: Assembler messages:
/home/kozmotronik/esp/ESP8266_RTOS_SDK/components/esp8266/include/driver/soc.h:103: Error: invalid register 'intclear' for 'wsr' instruction
make[1]: *** [/home/kozmotronik/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:292: source/task_wdt.o] Error 1
make: *** [/home/kozmotronik/esp/ESP8266_RTOS_SDK/make/project.mk:571: component-esp8266-build] Error 2
```

cango01, 2020-07-06:
i have same problem.
esp01 board. debian 10 os 
Power Supply: dc adp

nsfilho, 2020-07-18:
Are you using what version of  `toolchain`?

kozmotronik, 2020-07-18:
From the debug output:
Toolchain version: crosstool-ng-1.22.0-100-ge567ec7
Compiler version: 5.2.0

nsfilho, 2020-07-20:
Are you have cloned the repository in what way? What command are you used? Are you used recursive submodules?
Appears like your SDK aren't seeing `.git` folder. Are you used the official site tutorial for install your platform?

I remember of I see something exactly like that in my installation. But I don't remember the exactly steps to solve. I revised all steps in the official documentation about how to install and solve. But I don't remember exactly what step solved.

kozmotronik, 2020-07-20:
Mr. nsfilho, if you read the "**Problem Description**" section I have stated there that:  _I went step by step as per instructions in esp official docs: https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/#get-started-get-esp-id._ 
I used the Make to compile the Hello_World example. But somehow the compiler does not recognize a register of ESP8266 named: "intclear" with usage of wsr (assembly) instruction. I think it relates to a problem with paths or includes, otherwise it might be an error of the compiler toolchain.
I can understand that the developers are very busy but if they don't help I cannot overcome this issue by myself. Because the environment is designed by them.

nsfilho, 2020-07-21:
Sorry to not help you. As last try, I remembered what occured in my case, it's about the env variables (used by compiler).
Let me suggest you, before compile, clear some variables as:

```sh
    unset LDFLAGS
    unset CPPFLAGS
    unset XPC_FLAGS
```

I created a function to aliases all this things to me:

```sh
function idf8266()
{
    unset LDFLAGS
    unset CPPFLAGS
    unset XPC_FLAGS
    export IDF_PATH=/Users/neliosantos/workspace/esp/ESP8266_RTOS_SDK
    IDF_ADD_PATHS_EXTRAS="${IDF_PATH}/components/esptool_py/esptool"
	IDF_ADD_PATHS_EXTRAS="${IDF_ADD_PATHS_EXTRAS}:${IDF_PATH}/components/partition_table/"
	IDF_ADD_PATHS_EXTRAS="${IDF_ADD_PATHS_EXTRAS}:${IDF_PATH}/tools/"
    export PATH="$PATH:${IDF_ADD_PATHS_EXTRAS}:/Users/neliosantos/workspace/esp/xtensa-lx106-elf/bin"
	echo "Added to PATH: ${IDF_ADD_PATHS_EXTRAS}"
    if [ -e /dev/cu.SLAB_USBtoUART ] ; then
        export ESPPORT="/dev/cu.SLAB_USBtoUART"
    else
        export ESPPORT="`/bin/ls /dev/cu.usbserial-*`"
    fi
}
```

I hope this could help you 😄

nsfilho, 2020-07-23:
Hi @kozmotronik ! Are my last comment helped you to solve the problem? Are you fixed that? If yes, please share with the community the steps (this could help someone in future).

kozmotronik, 2020-07-28:
Hi @nsfilho, unfortunately because of my busy work I haven't been able to try it any more. I will try your suggestion soon and then I will post the result.

kozmotronik, 2023-08-20:
Hi guys @nsfilho @cango01! It's been 3 years and still no solution for this issue 🫣!!!... No worries fellas here I come with the solution after 3 years 🎁🎁🎁. I tried hard to find out what is the real cause of this issue. Because it didn't make sense to me that the compiler wouldn't recognize those assembly instructions, it was nonsense!  
I should confess that I have struggled a little to setup the enviorenment from scratch. This time my OS is OpenSUSE Tumbleweed KDE. Well after setting up the SDK, required python packages and the environment variables, it came to apply the solution so that the project is compiled successfully.  
### The Issue Was...
It turns out that the issue was a locale settings issue. My system's locale was set to `LANG=tr_TR.UTF-8` since I'am Turkish. This somehow was causing that Eclipse parse the source file with this locale and interpret some of the caracters different.
### The Solution Is...
Well after finding out what was the root cause of the problem, finding the solution is much easier. The solution is to launch Eclipse with the English locale settings as following:
```sh
LC_ALL=en_US.UTF-8 eclipse
```
However launching Eclipse from terminal in this way every time I need sucks. To make things easier, I modified the its shortcut .desktop file so that every time I launch it from the icon it sets up automatically. Just modify the `Exec` parameter as following and done.
```
Exec=**LC_ALL=en_US.UTF-8** /...path_to_eclipse.../eclipse
```
We've just added the `LC_ALL=en_US.UTF-8` statement with a space before the path and that's all. There we go, so I will close this issue as resolved.  
However I'd like to know whether the developers can deal with this locale problem in the source code files.

closed 2023-08-20 by kozmotronik