espressif/ESP8266_RTOS_SDK #932: 使用http server出现问题

GarMingLi, 2020-07-10:
我注册了一个 get方法的uri，用来发送网页数据给访问的client，但是老是出现错误：
```
W (17395) httpd_txrx: httpd_sock_err: error in send : 104
W (17403) httpd_txrx: httpd_sock_err: error in recv : 104
```
我的wifi模式时AP模式，断开AP再连接，重复几次，又会出现以下错误：
```
W (155851) httpd: httpd_accept_conn: error in accept (23)
W (155863) httpd: httpd_server: error accepting new connection
W (155918) httpd: httpd_accept_conn: error in accept (23)
W (155931) httpd: httpd_server: error accepting new connection
W (155966) httpd_txrx: httpd_sock_err: error in recv : 104
```
参数我已经设置很大了：
```
  httpd_config_t config = HTTPD_DEFAULT_CONFIG();
  config.max_open_sockets = 32;
  config.max_uri_handlers = 32;
  config.max_resp_headers = 32;
  config.backlog_conn = 32;
```

nsfilho, 2020-07-18:
Hi @GarMingLi !

I don't understand what you are saying, but I see the exactly same error in my project on http server. You can see my configurations if this help you.

[httpd.c](https://github.com/nsfilho/E12AIO3/blob/master/main/httpd.c) - of my project: https://github.com/nsfilho/E12AIO3

if you need any more informations, or if I could help you, please let  me know (but if you can write in english, help us to understanding what you are saying).

best regards, ns.

nsfilho, 2020-07-23:
Remember: the number of sockets opened depend not only from httpd_config_t, but you need configure lwIP to support it too and the use memory could be another important point too.

In my case, I need to serve many different HTMLs, SVGs, CSS, Javascript and etc., from my ESP8266 http server. What I do is process all (using GULP), mix all this data in 2 files and my `API` requests, I do one by one, in a promise way (to not flood ESP8266).

Well, I don't know if this your case. If you are providing sensor informations, HTTPD could be not the best way to do that, probably MQTT is better (one topic to many clients) and you can control the speed to send (or use a circular queue to enqueue all your datas to send).

Anyways, if you want to talk about your use case, we are here to help :)

GarMingLi, 2020-08-17:
> Remember: the number of sockets opened depend not only from httpd_config_t, but you need configure lwIP to support it too and the use memory could be another important point too.
>
> In my case, I need [...]
Nsfilho, thank you very much.I need to implement a webserver in my project.After the user connects to AP, the browser accesses a URL, and the Webserver accepts the request and returns HTML, JS, CSS and other files.And parse the associated POST and GET requests.I found that merging all the HTML, JS, and CSS files into one HTML file did not have this error problem.This error occurs as long as I don't merge the files and the browser accesses them only when it needs them.I don't know if you understand my description?

GarMingLi, 2020-08-17:
Hi,Nsfilho, thank you very much.
I need to implement a webserver in my project.After the user connects to AP, the browser accesses a URL, and the Webserver accepts the request and returns HTML, JS, CSS and other files.And parse the associated POST and GET requests.
I found that merging all the HTML, JS, and CSS files into one HTML file did not have this error problem.This error occurs as long as I don't merge the files and the browser accesses them only when it needs them.
I don't know if you understand my description?

GarMingLi, 2020-08-17:
> Remember: the number of sockets opened depend not only from httpd_config_t, but you need configure lwIP to support it too and the use memory could be another important point too.
>
> In my case, I need [...]

Well, thanks very much

closed 2020-08-17 by GarMingLi