espressif/ESP8266_RTOS_SDK #936: json中含有数字时,使用cJSON_Print()转字符串永远返回null

xizonghu, 2020-07-14:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at bbs.espressif.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:      [Arduino IDE|Make|Eclipse|other]
- Operating System:     [Windows|Ubuntu|MacOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

xizonghu, 2020-07-14:
cJSON *jsRes = cJSON_CreateObject();
    cJSON_AddStringToObject(jsRes, "ACK", "3424");
    cJSON_AddNumberToObject(jsRes, "ERRNO", 123);
    LOG_D(TAG, "[js: %08x]\n", jsRes);
    char *sss = cJSON_Print(jsRes);
    LOG_D(TAG, "[sss: %08x]\n", (u32)sss);
    LOG_D(TAG, "sss=%s\n", sss);
    free(sss);

nsfilho, 2020-07-18:
@xizonghu as a suggestion, you can see some `cjson` sample on my project (to compare the way as you are doing). In my `config.c` have a lot of ways I'm using and you could use in some of these ways.

[config.c](https://github.com/nsfilho/E12AIO3/blob/master/main/config.c) in project https://github.com/nsfilho/e12aio3

Best regards, NS.

berkutta, 2020-11-21:
Same issue here, `cJSON_Print` returns NULL:

```
    char *string = NULL;
    cJSON *monitor = cJSON_CreateObject();

    cJSON_AddNumberToObject(monitor, "cores", 5);

    string = cJSON_Print(monitor);
    if (string == NULL)
    {
        printf("Failed to print monitor.\n");
    } else {
        printf("%s", string);
    }

    cJSON_Delete(monitor);
```

berkutta, 2020-11-21:
On ESP32 this code works.

berkutta, 2020-11-22:
On ESP8266_RTOS_SDK V3.3 everything works too, on Master it returns NULL instead of a Pointer to a String. @donghengqaz @FayeY

berkutta, 2020-12-04:
It works for me when disabling `Enable 'nano' formatting options for printf/scanf family` in menuconfig. I think we can close this issue..

AFellmett, 2021-05-16:
> It works for me when disabling `Enable 'nano' formatting options for printf/scanf family` in menuconfig. I think we can close this issue..

Thx, this works fine for me. 
But added 10% extra size to my image just by disabling this option.
Would be great to have a fix using the `'nano ' formatting option`