espressif/ESP8266_RTOS_SDK #938: MQTT event_handler not receive more than 2 messages (short time)

nsfilho, 2020-07-21:
## Environment

- Development Kit: [ESP8266 Development Board](https://www.aliexpress.com/item/32991118728.html?spm=a2g0o.productlist.0.0.70067038ALowGS&algo_pvid=d2fa122b-11a0-4975-b5b2-bab46eb7abb0&algo_expid=d2fa122b-11a0-4975-b5b2-bab46eb7abb0-8&btsid=0ab6f83915953512203374296e6fb4&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)
- IDF version: be7e3fd7
- Development Env: Visual Studio Code
- Operating System: MacOS
- Power Supply: USB

## Problem Description

Using [esp-mqtt sample](https://github.com/espressif/ESP8266_RTOS_SDK/tree/release/v3.3/examples/protocols/esp-mqtt/tcp), when I send more than 2 messages quickly, we will not receive others. Independent what QoS I use.

My real use case [firmware](https://github.com/nsfilho/e12aio3), I receive 3 messages quickly to active some pins (relays), but the ESP only receive 2 of them (in a short time), if I send one by one, I receive all messages.

When I used PubSubClient (in Arduino emulation mode), I receive all of them. Because of that, I know this isn't a hardware limitation, but probably about the implementation.

**MQTT Server Version**: 3.1.1 (mosquitto)
**Other variables**: without SSL, pure TCP (port: 1883).

### Expected Behavior

Receive all messages from MQTT server.

### Actual Behavior

Receive only the first 2 messages.

### Steps to repropduce

1. Configure the sample code [esp-mqtt sample](https://github.com/espressif/ESP8266_RTOS_SDK/tree/release/v3.3/examples/protocols/esp-mqtt/tcp);
2. Try to send 3, 4 or 5 messages to `/topc/qos0` (for example);

> Important: use a simple software to send that messages without needs to login all the times (i.e. node-red with inject), because the time to login makes the send messages comes with some delay.

[image]

nsfilho, 2020-07-21:
@donghengqaz please help me 😅

cranphin, 2020-08-04:
Is this the same as https://github.com/espressif/ESP8266_RTOS_SDK/issues/907 ?

nsfilho, 2020-08-08:
@cranphin yes, it is the exactly same case. Now I have implemented a new stack of esp-mqtt in my project too. It is fully functional (receiving multiples messages at the same time).

Thank you by your support. My unique concern it is about EspressIf, because no one answered. The Espressif agenda, isn't concern about the developers (what IMHO is an error).

Well, anyways, if you have the same problem (or other people), I'm here to help in anything of I can. Including showing how to implement that stack. OK?

closed 2020-08-08 by nsfilho