espressif/ESP8266_RTOS_SDK #940: Error with Python

ivanphidalgo, 2020-07-25:
I'm having issues with make menuconfig. This is the error:

`pkg_resources cannot be imported probably because the pip package is not installed and/or using a legacy Python interpreter. Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
make: *** No rule to make target 'check_python_dependencies', needed by 'menuconfig'.  Stop.`

Any help?

MrHause, 2020-08-18:
Hi, 
which operating system do u use?
Did you install the requirements included in the SDK?
[Requirements](https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/#install-the-required-python-packages)
Check also whether your environment variables IDF_PATH and PATH is set properly.

midnight-wonderer, 2020-09-25:
I've got the same error.

For me, I have `pip` command referred to `pip` of python3 in my user directory.
However, `python` command is an alias of python2 because I am on a Debian-based distro, and aliasing `python` command to python3 would cause a lot of trouble. Hence, the error.

The workaround for me is to run menuconfig with `make menuconfig CONFIG_SDK_PYTHON=python3`.

Arian94, 2023-03-26:
After that, go to ``` SDK tool configuration```>```Python 2 interpreter```, hit ```enter``` and set it to ```python3```. Save and close.

Now use ```make menuconfig``` as usual.

> I've got the same error.
>
> For me, I have `pip` command referred to `pip` of python3 in my user directory. However, `python` command is an alias of python2 because I am on a Debian-based distro, and [...]