espressif/ESP8266_RTOS_SDK #941: 刚刚pull了最新主干上的代码后，程序跑起来后无法连接到wifi

microponts, 2020-07-28:
RT,以及mqtt不能发送大数据量(1k以内)的消息

huozhe0930, 2020-08-16:
同问，最新SDK，无法连接到wifi。没有更新SDK前一切正常。更新后无法连接。用smart_config example测试，也不行。
错误信息如下：
I (6678) APP_MAIN: SC_STATUS_GETTING_SSID_PSWD
T|pswd: ***
T|ssid: ***
I (10108) APP_MAIN: SC_STATUS_LINK
I (10111) APP_MAIN: SSID:***
I (10114) APP_MAIN: PASSWORD:***
I (13288) APP_MAIN: ip connect is lost!, trying to connect again...

I (16458) APP_MAIN: ip connect is lost!, trying to connect again...

I (19629) APP_MAIN: ip connect is lost!, trying to connect again...

I (22799) APP_MAIN: ip connect is lost!, trying to connect again...

I (25970) APP_MAIN: ip connect is lost!, trying to connect again...

I (29141) APP_MAIN: ip connect is lost!, trying to connect again...

I (32311) APP_MAIN: ip connect is lost!, trying to connect again...

I (35482) APP_MAIN: ip connect is lost!, trying to connect again...

I (38653) APP_MAIN: ip connect is lost!, trying to connect again...

I (41823) APP_MAIN: ip connect is lost!, trying to connect again...

I (44994) APP_MAIN: ip connect is lost!, trying to connect again...

I (48164) APP_MAIN: ip connect is lost!, trying to connect again...

I (51335) APP_MAIN: ip connect is lost!, trying to connect again...

I (54506) APP_MAIN: ip connect is lost!, trying to connect again...

I (57677) APP_MAIN: ip connect is lost!, trying to connect again...

I (60848) APP_MAIN: ip connect is lost!, trying to connect again...

I (64018) APP_MAIN: ip connect is lost!, trying to connect again...

I (64023) APP_MAIN:
You should enter wrong password, clear ssid config and reset!

microponts, 2020-08-16:
主干上的代码估计有问题，可以回退到3.3v的版本

mosajon, 2020-09-01:
此更新导致的：
https://github.com/espressif/ESP8266_RTOS_SDK/commit/321c160529f9b20777421a4adcbd39250b425c10