espressif/ESP8266_RTOS_SDK #942: WiFi broken on Master fe99ec4

berkutta, 2020-07-28:
It connects to wifi but IP Adress event is not there. 

```
--- idf_monitor on /dev/ttyUSB2 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
aag
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7168, room 16 
tail 0
chksum 0x03
load 0x3ffe8408, len 24, room 8 
tail 0
chksum 0x8a
load 0x3ffe8420, len 3532, room 8 
tail 4
chksum 0x93
I (44) boot: ESP-IDF v3.2-526-gfe99ec40 2nd stage bootloader
I (44) boot: compile time 17:23:56
I (44) qio_mode: Enabling default flash chip QIO
I (52) boot: SPI Speed      : 40MHz
I (59) boot: SPI Mode       : QIO
I (65) boot: SPI Flash Size : 2MB
I (71) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (88) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (99) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (111) boot:  2 factory          factory app      00 00 00010000 000f0000
I (122) boot: End of partition table
I (129) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x51cf4 (335092) map
0x40210010: _stext at ??:?

I (255) esp_image: segment 1: paddr=0x00061d0c vaddr=0x40261d04 size=0x0dff8 ( 57336) map
I (274) esp_image: segment 2: paddr=0x0006fd0c vaddr=0x3ffe8000 size=0x006d0 (  1744) load
I (276) esp_image: segment 3: paddr=0x000703e4 vaddr=0x40100000 size=0x00bd0 (  3024) load
I (286) esp_image: segment 4: paddr=0x00070fbc vaddr=0x40100bd0 size=0x058b4 ( 22708) load
I (306) boot: Loaded app from partition at offset 0x10000
I (326) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (332) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (350) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
I (350) wifi station: ESP_WIFI_MODE_STA
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
I (407) phy_init: phy ver: 1163_0
I (438) wifi station: wifi_init_sta finished.
I (577) wifi: state: 0 -> 2 (b0)
I (584) wifi: state: 2 -> 3 (0)
I (591) wifi: state: 3 -> 5 (10)

```

cranphin, 2020-07-29:
I think I'm seeing the same, sometimes it works, but usually it fails.
Switching back to f7e1bb83fb2d2c72ebcd2e72c8c88849a5a8e31f seemt to fix it.

Bad: (v3.2-526-g**fe99ec40**)
```
(45) boot: ESP-IDF v3.2-526-gfe99ec40 2nd stage bootloader
(45) boot: compile time 10:52:54
(45) qio_mode: Enabling default flash chip QIO
(50) boot: SPI Speed      : 40MHz
(54) boot: SPI Mode       : QIO
(58) boot: SPI Flash Size : 4MB
(62) boot: Partition Table:
(66) boot: ## Label            Usage          Type ST Offset   Length
(73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
(80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
(88) boot:  2 factory          factory app      00 00 00010000 000f0000
(95) boot: End of partition table
(99) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x59cbc (367804) map
(232) esp_image: segment 1: paddr=0x00069cd4 vaddr=0x40269ccc size=0x0eed8 ( 61144) map
(253) esp_image: segment 2: paddr=0x00078bb4 vaddr=0x3ffe8000 size=0x00730 (  1840) load
(254) esp_image: segment 3: paddr=0x000792ec vaddr=0x40100000 size=0x00bd0 (  3024) load
(261) esp_image: segment 4: paddr=0x00079ec4 vaddr=0x40100bd0 size=0x05bac ( 23468) load
(277) boot: Loaded app from partition at offset 0x10000
(303) system_api: Base MAC address is not set, read default base MAC address from EFUSE
(311) system_api: Base MAC address is not set, read default base MAC address from EFUSE
(324) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
(329) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(335) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(342) main: Free memory: 95964 bytes
(347) main: IDF version: v3.2-526-gfe99ec40
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
(400) phy_init: phy ver: 1163_0
(415) app_wifi: wifi_init_sta finished.
(418) main: Waiting for WiFi
(422) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(1490) wifi: state: 0 -> 2 (b0)
(1498) wifi: state: 2 -> 3 (0)
(1504) wifi: state: 3 -> 5 (10)
```

Also bad: (v3.2-523-g**00fafb8d**)
```
(45) boot: ESP-IDF v3.2-523-g00fafb8d 2nd stage bootloader
(45) boot: compile time 10:42:16
(45) qio_mode: Enabling default flash chip QIO
(50) boot: SPI Speed      : 40MHz
(54) boot: SPI Mode       : QIO
(58) boot: SPI Flash Size : 4MB
(62) boot: Partition Table:
(66) boot: ## Label            Usage          Type ST Offset   Length
(73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
(80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
(88) boot:  2 factory          factory app      00 00 00010000 000f0000
(95) boot: End of partition table
(99) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x59cbc (367804) map
(232) esp_image: segment 1: paddr=0x00069cd4 vaddr=0x40269ccc size=0x0eed8 ( 61144) map
(253) esp_image: segment 2: paddr=0x00078bb4 vaddr=0x3ffe8000 size=0x00730 (  1840) load
(254) esp_image: segment 3: paddr=0x000792ec vaddr=0x40100000 size=0x00bd0 (  3024) load
(261) esp_image: segment 4: paddr=0x00079ec4 vaddr=0x40100bd0 size=0x05bfc ( 23548) load
(277) boot: Loaded app from partition at offset 0x10000
(303) system_api: Base MAC address is not set, read default base MAC address from EFUSE
(311) system_api: Base MAC address is not set, read default base MAC address from EFUSE
(324) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
(329) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(335) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(342) main: Free memory: 95884 bytes
(347) main: IDF version: v3.2-523-g00fafb8d
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
(401) phy_init: phy ver: 1163_0
(416) app_wifi: wifi_init_sta finished.
[... 25 lines cut ...]
(261) esp_image: segment 4: paddr=0x000798ec vaddr=0x40100bd0 size=0x05b7c ( 23420) load
(277) boot: Loaded app from partition at offset 0x10000
(308) system_api: Base MAC address is not set, read default base MAC address from EFUSE
(316) system_api: Base MAC address is not set, read default base MAC address from EFUSE
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
(376) phy_init: phy ver: 1163_0
(380) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
(386) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(393) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(400) main: Free memory: 96084 bytes
(404) main: IDF version: v3.2-504-gf7e1bb83-dirty
(429) app_wifi: wifi_init_sta finished.
(432) main: Waiting for WiFi
(438) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(1567) wifi: state: 0 -> 2 (b0)
(1573) wifi: state: 2 -> 3 (0)
(1581) wifi: state: 3 -> 5 (10)
(1582) wifi: pm start, type: 1
(2571) tcpip_adapter: sta ip: 192.168.178.45, mask: 255.255.255.0, gw: 192.168.178.1
(2578) app_wifi: got ip:192.168.178.45
(2581) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(2589) doorbell: Initialize hw_timer for transmitter_callback
(2596) doorbell: Set hw_timer timing time 0.5ms with reload
(2602) main: Start button timer
```

cranphin, 2020-07-29:
And another example run on f7e1bb83, this time not dirty :)
So:
- fe99ec4: bad
- 00fafb8: bad
- f7e1bb8: good
I did not test the versions inbetween, it's very time consuming :)

```
(45) boot: ESP-IDF v3.2-504-gf7e1bb83 2nd stage bootloader
(45) boot: compile time 11:06:43
(45) qio_mode: Enabling default flash chip QIO
(50) boot: SPI Speed      : 40MHz
(54) boot: SPI Mode       : QIO
(58) boot: SPI Flash Size : 4MB
(62) boot: Partition Table:
(66) boot: ## Label            Usage          Type ST Offset   Length
(73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
(80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
(88) boot:  2 factory          factory app      00 00 00010000 000f0000
(95) boot: End of partition table
(99) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x58e14 (364052) map
(230) esp_image: segment 1: paddr=0x00068e2c vaddr=0x40268e24 size=0x0e778 ( 59256) map
(251) esp_image: segment 2: paddr=0x000775ac vaddr=0x3ffe8000 size=0x006f0 (  1776) load
(252) esp_image: segment 3: paddr=0x00077ca4 vaddr=0x40100000 size=0x00bd0 (  3024) load
(259) esp_image: segment 4: paddr=0x0007887c vaddr=0x40100bd0 size=0x05b7c ( 23420) load
(275) boot: Loaded app from partition at offset 0x10000
(300) system_api: Base MAC address is not set, read default base MAC address from EFUSE
(309) system_api: Base MAC address is not set, read default base MAC address from EFUSE
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
(368) phy_init: phy ver: 1163_0
(372) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
(378) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(387) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(395) main: Free memory: 96116 bytes
(400) main: IDF version: v3.2-504-gf7e1bb83
(422) app_wifi: wifi_init_sta finished.
(427) main: Waiting for WiFi
(434) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(2431) wifi: state: 0 -> 2 (b0)
(2438) wifi: state: 2 -> 3 (0)
(2443) wifi: state: 3 -> 5 (10)
(2445) wifi: pm start, type: 1
(3440) tcpip_adapter: sta ip: 192.168.178.45, mask: 255.255.255.0, gw: 192.168.178.1
(3449) app_wifi: got ip:192.168.178.45
(3453) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(3462) doorbell: Initialize hw_timer for transmitter_callback
(3468) doorbell: Set hw_timer timing time 0.5ms with reload
(3474) main: Start button timer
```

berkutta, 2020-08-12:
Seems to work again with latest commit

```
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
I (408) phy_init: phy ver: 1163_0
I (443) wifi station: wifi_init_sta finished.
I (566) wifi: state: 0 -> 2 (b0)
I (575) wifi: state: 2 -> 3 (0)
I (580) wifi: state: 3 -> 5 (10)
I (2843) tcpip_adapter: sta ip: 192.168.0.236, mask: 255.255.255.0, gw: 192.168.0.1
I (2847) wifi station: got ip:192.168.0.236
```

closed 2020-08-12 by berkutta

cranphin, 2020-08-12:
Yup, seems good for me too sofar