espressif/ESP8266_RTOS_SDK #945: Can't build after upgrading to commit 5dbf32a928254e97ede794ac47fd53b80b66e5ed or higher

d51x, 2020-07-30:
Master branch.
Commit  741baf8a3211c6d5f25d54ed6b09be302a632235.
It is possible to build project a project.

After upgrade to commit 5dbf32a928254e97ede794ac47fd53b80b66e5ed or higher I can't build the project.

`
ESP8266_RTOS_SDK/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c: In function 'crypto_ec_set_pubkey_point':
ESP8266_RTOS_SDK/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:531:20: error: implicit declaration of function 'mbedtls_pk_ec' [-Werror=implicit-function-declaration]
  mbedtls_ecp_copy(&mbedtls_pk_ec(*key)->Q, point);

ESP8266_RTOS_SDK/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:531:39: error: invalid type argument of '->' (have 'int')
  mbedtls_ecp_copy(&mbedtls_pk_ec(*key)->Q, point);

ESP8266_RTOS_SDK/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:532:45: error: invalid type argument of '->' (have 'int')
  mbedtls_ecp_group_load(&mbedtls_pk_ec(*key)->grp, MBEDTLS_ECP_DP_SECP256R1);

ESP8266_RTOS_SDK/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c: In function 'crypto_ec_get_public_key':
ESP8266_RTOS_SDK/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:559:56: error: invalid type argument of '->' (have 'int')
  return (struct crypto_ec_point *)&mbedtls_pk_ec(*pkey)->Q;
................
..... e.t.c....
................
ESP8266_RTOS_SDK/components/wpa_supplicant/src/crypto/crypto_mbedtls-ec.c:753:2: error: unknown type name 'mbedtls_ecdsa_context'
  mbedtls_ecdsa_context *ctx = os_malloc(sizeof(*ctx));
............
...........
/ESP8266_RTOS_SDK/make/component_wrapper.mk:291: recipe for target 'src/crypto/crypto_mbedtls-ec.o' failed
make[1]: *** [src/crypto/crypto_mbedtls-ec.o] Error 1
make[1]: *** Waiting for unfinished jobs....
/ESP8266_RTOS_SDK/make/project.mk:571: recipe for target 'component-wpa_supplicant-build' failed
make: *** [component-wpa_supplicant-build] Error 2
`

I tried to build with CONFIG_ESP8266_WIFI_ENABLE_WPA3_SAE = y and CONFIG_ESP8266_WIFI_ENABLE_WPA3_SAE = n in sdkconfig.

d51x, 2020-07-31:
SOLVED,  my mistake, incorrect settings in sdkconfig

closed 2020-07-31 by d51x

zheyuejiang, 2020-11-14:
I met the same problem, how to change some configs?

optionalis, 2020-12-08:
> I met the same problem, how to change some configs?

> I met the same problem, how to change some configs?

我也遇到相同的问题，你那边解决了嘛。

dsptech, 2020-12-11:
Hi, enabling elliptic curve ciphers in MBEDTLS solved the issue for me.