espressif/ESP8266_RTOS_SDK #946: I2C Issue with GPIO 4 & 5

berkutta, 2020-07-30:
Are there any known issues with GPIO 4 & 5, I2C, v3.3 RTOS SDK and HTU21D Sensor known?

If I use the following configuration `sda_io_num = 4, scl_io_num = 5` I can hook up a Logic Analyzer or Oscilloscope and reading from the HTU21D Sensor fails. The HTU21D doesn't ACK any more on write or read. Also touching the pins with my hand results in failed sensor readings. 

When using `sda_io_num = 14, scl_io_num = 2` I don't get any issues with touching the pins or hooking up a Logic Analyzer.

anticodeninja, 2021-05-06:
Thanks @berkutta, I have the same issue and this info has saved a lot of time on debugging. And the issue still exists in 911e2d7a. Also everything works fine with GPIO 4&5 on Arduino with Wire library, so it looks like the issue in SDK, but I haven't found a way to fix it or even understand why it happens.

dsptech, 2021-05-06:
Hi, this PR may solve your issue:
https://github.com/espressif/ESP8266_RTOS_SDK/pull/1071#issue-602518466