espressif/ESP8266_RTOS_SDK #952: 如何使用ESP8266 LWIP中的NAT功能实现WiFi中继呢？？

GIT111222, 2020-08-17:
如何使用ESP8266 LWIP中的NAT功能实现WiFi中继呢？？

vinhjaxt, 2020-09-06:
Follow to: https://github.com/martin-ger/esp32_nat_router
```
make menuconfig
--> Component config --> LWIP:
[*] Enable copy between Layer2 and Layer3 packets
[*] Enable IP forwarding
[*]   Enable NAT (new/experimental)
```
The code:
```c
#include "lwip/opt.h"
#include "lwip/ip4_napt.h"
#include "tcpip_adapter.h"
#include "esp_netif.h"
#include "lwip/err.h"
#include "lwip/sys.h"

    esp_netif_init();
    u32_t napt_netif_ip = 0xC0A80401; // Set to ip address of softAP netif (Default is 192.168.4.1)
    ip_napt_enable(htonl(napt_netif_ip), 1);
```
But not work yet. Packet went out, and never come in :(

vinhjaxt, 2020-09-06:
Local LAN still works :/