espressif/ESP8266_RTOS_SDK #962: Firmware too large problem (IDFGH-3937)

GarMingLi, 2020-09-07:
I used ESP8266 to implement a WebServer. 900KB web page file in the firmware, and I used the **COMPONENT_EMBED_TXTFILES** method to compile the file into the firmware.Because the program is too large, I have redefined the partition.
This is the contents of my partition file:
```
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,16K,
otadata,data,ota,0xd000,8K,
phy_init,data,phy,0xf000,4K,
ota_0,app,ota_0,0x10000,2000K,
ota_1,app,ota_1,0x208000,2000K,
```
After changing the partition, I can compile successfully.But after I downloaded and ran it, there was an error:
```I (315) esp_image: segment 1: paddr=0x0008104c vaddr=0x40281044 size=0xb9db4 (761268) map
I (635) esp_image: segment 2: paddr=0x0013ae08 vaddr=0x3ffe8000 size=0x00628 (  1576) load
I (636) esp_image: segment 3: paddr=0x0013b438 vaddr=0x40100000 size=0x00a50 (  2640) load
I (643) esp_image: segment 4: paddr=0x0013be90 vaddr=0x40100a50 size=0x05a48 ( 23112) load
I (660) boot: Loaded app from partition at offset 0x10000
E (660) boot: ERROR: app bin error, start_addr 10000 image_len 1251552
```
According to the error prompt, I went to find the source code,In the "bootloader_utility_load_image" function:
```
#define ESP_CACHE1_ADDR_MAX 0x100000
#define ESP_CACHE2_ADDR_MAX 0x200000

    size_t map;

    if (image_data->start_addr < ESP_CACHE1_ADDR_MAX
        && image_data->start_addr + image_data->image_len < ESP_CACHE1_ADDR_MAX) { 
        map = 0;
    } else if (image_data->start_addr >= ESP_CACHE1_ADDR_MAX
               && image_data->start_addr < ESP_CACHE2_ADDR_MAX
               && image_data->start_addr + image_data->image_len < ESP_CACHE2_ADDR_MAX) {
        map = 1;
    } else {
        ESP_LOGE(TAG, "ERROR: app bin error, start_addr %x image_len %d\n", image_data->start_addr, image_data->image_len);
        /* Blocking here to let user judge. */
        while (1);
    }

    Cache_Read_Enable(map, 0, SOC_CACHE_SIZE);

    user_start = (void *)image_data->image.entry_addr;
    user_start(image_data->start_addr);
```
This means that if my firmware ends up exceeding 1M, it won't boot.

wujiangang, 2020-09-07:
For ESP8266, ota_0 should be within the first 1MB, so it starts from 0x10000, and size is 0xF0000, ota_1 shoud be with the second 1MB, it starts from 0x110000, size is 0xF0000.

In your case, you can put your webserver in a seperate partition, the partition is such as:
```
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,16K,
otadata,data,ota,0xd000,8K,
phy_init,data,phy,0xf000,4K,
ota_0,app,ota_0,0x10000,0xF0000,
ota_1,app,ota_1,0x110000,0xF0000,
storage,    data, ,             , 0x200000,
```

GarMingLi, 2020-09-07:
In 8266, even though its Flash size is 4M, can't the app firmware be larger than 1M?
Then I wouldn't have been able to compile the file into the firmware using the COMPONENT_EMBED_TXTFILES method. I would have had to put the file in the last two meters of space.
@wujiangang

whwxxstsyd, 2020-12-29:
@GarMingLi 
have you already solved the problem yet?
I got the same like you do.
maybe you can try as follows

1.make erase_flash(make sure your flash is clean, very important)
2.make menuconfig , Serial flasher config  ---> Flash size  (select the correct Flash size)

after this two steps, I can successfully run

[image]

[image]

zheyuejiang, 2020-12-29:
do you have try single  partition to compile. The 2M maybe too large for esp8266. you can try esp32S . @GarMingLi

GarMingLi, 2021-01-26:
> do you have try single partition to compile. The 2M maybe too large for esp8266. you can try esp32S . @GarMingLi

试过了，一样的

closed 2023-08-28 by wujiangang