espressif/ESP8266_RTOS_SDK #971: os_timer_arm_us  alternative

hiperiondev, 2020-09-16:
I need a timer call at uS resolution.

Software timer not allow this resolution and hardware timer makes crazy things with FreeRTOS.

os_timer_arm_us is not supported by ESP8266_RTOS_SDK .

There is an alternative?

dsptech, 2020-09-17:
Hi,
what do you mean with "... hardware timer makes crazy things with FreeRTOS" ?

The us resolution is a bit hard due the continuous (heavy) interference from the wifi driver, but a solution could exist. Could you describe better your needs ? 

Regards.

hiperiondev, 2020-09-17:
I'm trying to port freemodbus for ESP8266_RTOS_SDK. This need a timer with a granularity of 50 uS.
If I use hw timer wifi some times not work and rtos timing is not stable.

dsptech, 2020-09-17:
I think that you are experiencing two kind of "non timer related issues". 

- - - - - -
The first issue is that the wifi driver takes possession of the cpu by disabling interrupts for a very long time. 

See this two issues for info:
https://github.com/espressif/ESP8266_RTOS_SDK/issues/626#issue-455083710
https://github.com/espressif/ESP8266_RTOS_SDK/issues/240#issue-335506136
As third case, the WPS can disable all processes and interrupts up to 7 seconds (@ 80Mhz) when negotiating.

The only workaround I found is to use the NMI by setting the wdev counter (see the the pwm driver for reference) and avoid the use of any freertos API from the handler.
The hardware timer can also be used to generate NMI, but currently I am not aware that it is supported by this sdk (this feature is supported in NONOS sdk instead).
- - - - - 
The second issue is that you need to leave enough cpu time to allow the work of the wifi driver. I've no precise data about that.
As example, In a my library I use a simple handler (NMI) to generate a tone with precise frequency on a pin. The handler is very light (more than the PWM driver) but if I set frequencies >40Khz (@80Mhz clock), sometime the wifi (station mode) fail to connect to the AP.

I hope this information will be useful to you