espressif/ESP8266_RTOS_SDK #974: Addition of the espressif/libcoap (9ed7356f) results in 404 while trying to update submodules.

sobomax, 2020-09-21:
Hi, JFYI: the espressif/libcoap  added in rev 9ed7356f into submodules results in 404 on GitHub. Open it in your browser and see by yourself. https://github.com/espressif/libcoap/

Please fix. Thanks!

-Max

```
hudson.plugins.git.GitException: Command "git submodule update --init --recursive components/coap/libcoap" returned status code 1:
stdout: 
stderr: Cloning into '/home/sobomax/jenkins/workspace/microsippy/ESP8266_RTOS_SDK/components/coap/libcoap'...
remote: Invalid username or password.
fatal: Authentication failed for 'https://github.com/espressif/libcoap/'
fatal: clone of 'https://github.com/espressif/libcoap' into submodule path '/home/sobomax/jenkins/workspace/microsippy/ESP8266_RTOS_SDK/components/coap/libcoap' failed
Failed to clone 'components/coap/libcoap'. Retry scheduled
Cloning into '/home/sobomax/jenkins/workspace/microsippy/ESP8266_RTOS_SDK/components/coap/libcoap'...
remote: Invalid username or password.
fatal: Authentication failed for 'https://github.com/espressif/libcoap/'
fatal: clone of 'https://github.com/espressif/libcoap' into submodule path '/home/sobomax/jenkins/workspace/microsippy/ESP8266_RTOS_SDK/components/coap/libcoap' failed
Failed to clone 'components/coap/libcoap' a second time, aborting
```

CIPop, 2020-09-21:
I'm not able to build an application with latest code (getting multiple issues saying lwip.h and mbedtls headers are missing). 
As workaround I was able to use the v3.3 release.

sobomax, 2020-10-16:
Seems to be fixed lately.

closed 2020-10-16 by sobomax