espressif/ESP8266_RTOS_SDK #981: I2S driver doesn't work

QB4-dev, 2020-09-28:
## Environment

- Development Kit: NodeMCU + PCM1502A I2S DAC
- IDF version (``git rev-parse --short HEAD`` to get the commit id.):  #917 (release/v3.3)  as well as master
    //483a5a475c7afdf3336c66fa279c70de4bb8b750
- Development Env:      Eclipse
- Operating System:     Ubuntu
- Power Supply:            USB + external 5V

## Problem Description

I have no output from on I2S. I'am using official example:
https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/peripherals/i2s 

Using `i2s_write( )` function is hanging until timeout occurs without writing data into DMA Queue
https://github.com/espressif/ESP8266_RTOS_SDK/blob/cbfad90f6d87b301dc6e3fe04c8f0acfdcfb4de3/examples/peripherals/i2s/main/i2s_example_main.c#L68

Every time `i2s_bytes_write` is 0

### Expected Behavior

Triangle and sine 100Hz wave on I2S DAC output

### Actual Behavior

When Debug message level is set to INFO, serial monitor output is as expected:

```
Test bits=24 free mem=78188, written data=2880
I (5655) i2s: DMA Malloc info, datalen=blocksize=480, dma_buf_count=6

Test bits=16 free mem=78188, written data=1440
I (10735) i2s: DMA Malloc info, datalen=blocksize=240, dma_buf_count=6

Test bits=24 free mem=78188, written data=2880
I (15835) i2s: DMA Malloc info, datalen=blocksize=480, dma_buf_count=6

Test bits=16 free mem=78188, written data=1440
I (20915) i2s: DMA Malloc info, datalen=blocksize=240, dma_buf_count=6
```

but only reason why in this example is running is `break` on timeout in `xQueueReceive'

https://github.com/espressif/ESP8266_RTOS_SDK/blob/cbfad90f6d87b301dc6e3fe04c8f0acfdcfb4de3/components/esp8266/driver/i2s.c#L748-L771

If You pass `portMAX_DELAY` as `ticks_to_wait` function `i2s_bytes_write` will hang forever

When DEBUG info is enabled there should be info about samples write 
https://github.com/espressif/ESP8266_RTOS_SDK/blob/cbfad90f6d87b301dc6e3fe04c8f0acfdcfb4de3/components/esp8266/driver/i2s.c#L757

like :

```
D (20915) i2s: size: 256, rw_pos: 16 , buf_size: 256, curr_ptr: 256

```
but there is no output like that.

### Steps to repropduce

1. Compile and run i2s example: https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/peripherals/i2s
2. Set log verbosity to DEBUG
3. Run again 'monitor` with debug info: no DMA write

doragasu, 2020-10-28:
I can confirm this problem on a nodemcu, branch is `release/v3.3`, commit `210943a9`

Please provide a fix or workaround.

doragasu, 2020-10-28:
Tried also with latest `release/v3.3` and latest master (with gcc 8.2) and the problem persists.

doragasu, 2020-10-29:
I have plugged a scope and it seems code is working! The problem is that on the nodemcu, the GPIO3 pin is connected to the CP2102 TX pin through a 470 ohm resistor, and it is trying to pull the pin high. This causes problems on this specific board. We would need to test this on a different board that has this pin unconnected, or at least not connected to another output.

QB4-dev, 2020-12-01:
@doragasu 
Yup. You are right, when DAC is connected to NodeMCU I2S works without issues but if logic analyzer only is connected GPIO3 line is pulled to high. So I can confirm that this is board connected issue.

closed 2020-12-01 by QB4-dev

flytobluesky, 2021-06-28:
I can confirm this problem on a nodemcu, branch is release/v3.4   ,  the 470 ohm  resistors between  ESP12E and cp2102 is removed, I2S is still not working properly.   GPIO2 GPIO3 and GPIO15 have no wave.

Steps to repropduce
Compile and run i2s example: https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/peripherals/i2s

papadeltasierra, 2022-08-07:
I'm not seeing this work with an ESP8266-01 device either.  The "bytes_write" value is always zero.  DMA does work on this device because I've previously (3-4 years ago?) had this working having reworked the old freertos code myself.

papadeltasierra, 2022-08-07:
As this appears to be closed, I'm going to open a new issue, referencing this one.