espressif/ESP8266_RTOS_SDK #991: shared_ptr copy constructor crashes in the latest toolchain (gcc 8.4)

Crspy, 2020-10-18:
## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:     Eclipse
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

//Detailed problem description goes here.

### Expected Behavior
it shouldn't crash when copy constructing a shared_ptr

### Actual Behavior
it crashes when copy constructing a shared_ptr using the latest toolchain (gcc 8.4)

### Steps to repropduce

1. just trying copy constructing any shared_ptr

### Code to reproduce this issue

```cpp
#include "esp_log.h"

struct test
{
	int x;
	int y;
};
std::shared_ptr<test> func()
{
	auto t = std::make_shared<test>();
	ESP_LOGI(TAG,"TEST"\n");
	return t;
}

extern "C" void app_main()
{
        auto t = func();
        ESP_LOGI(TAG,"%d\n",t->x);
}

```

## Debug Logs

```
abort() was called at PC 0x4025a1d5 on core 0
0x4025a1d5: uw_init_context_1 at /home/wjg/Repo/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/libgcc/unwind-dw2.c:1567
(discriminator 1)

Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x402270d1  PS      : 0x00000030  A0      : 0x402270cc  A1      : 0x3ffeba20
0x402270d1: abort at C:/esp/ESP8266_RTOS_SDK/components/newlib/src/syscall.c:69

0x402270cc: abort at C:/esp/ESP8266_RTOS_SDK/components/newlib/src/syscall.c:69

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000001
A6      : 0x4026cbd8  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x3ffebc44  A13     : 0x4025a739
0x4025a739: _Unwind_RaiseException at /home/wjg/Repo/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/libgcc/unwind.inc:89

A14     : 0x40234ea2  A15     : 0x3ffebce0  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d
0x40234ea2: __cxa_throw at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-lx106-elf/src/gcc/libstdc++-v3/l
ibsupc++/eh_throw.cc:94

Backtrace: 0x402270d1:0x3ffeba20 0x4025a1d8:0x3ffeba30 0x4025a739:0x3ffebb10 Guru Meditation Error: Core  0 panic'ed (Lo
adStoreAlignment). Exception was unhandled.
0x402270d1: abort at C:/esp/ESP8266_RTOS_SDK/components/newlib/src/syscall.c:69

0x4025a1d8: uw_init_context_1 at /home/wjg/Repo/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/libgcc/unwind-dw2.c:1577

0x4025a739: _Unwind_RaiseException at /home/wjg/Repo/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/libgcc/unwind.inc:89

Core 0 register dump:
PC      : 0x4023744f  PS      : 0x00000033  A0      : 0x402373c9  A1      : 0x3ffea110
0x4023744f: xt_retaddr_callee at C:/esp/ESP8266_RTOS_SDK/components/esp8266/source/backtrace.c:83

0x402373c9: xt_retaddr_callee at C:/esp/ESP8266_RTOS_SDK/components/esp8266/source/backtrace.c:50

A2      : 0x3ffebbdd  A3      : 0x000000c0  A4      : 0x000000d0  A5      : 0x000000a0
A6      : 0x3ffea140  A7      : 0x4025a708  A8      : 0x3ffebbe1  A9      : 0x000007fe
0x4025a708: _Unwind_RaiseException at /home/wjg/Repo/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/libgcc/unwind.inc:83

A10     : 0x00000090  A11     : 0x00000011  A12     : 0x402270cc  A13     : 0x3ffea150
0x402270cc: abort at C:/esp/ESP8266_RTOS_SDK/components/newlib/src/syscall.c:69

A14     : 0x4025a739  A15     : 0x3ffebb10  SAR     : 0x0000001f  EXCCAUSE: 0x00000009
0x4025a739: _Unwind_RaiseException at /home/wjg/Repo/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/libgcc/unwind.inc:89

Backtrace: 0x4023744f:0x3ffea110 0x4021fc3b:0x3ffea140 0x40100be0:0x3ffea1a0
0x4023744f: xt_retaddr_callee at C:/esp/ESP8266_RTOS_SDK/components/esp8266/source/backtrace.c:83

0x4021fc3b: panic_frame at C:/esp/ESP8266_RTOS_SDK/components/freertos/port/esp8266/panic.c:109
 (inlined by) panicHandler at C:/esp/ESP8266_RTOS_SDK/components/freertos/port/esp8266/panic.c:168

0x40100be0: _xt_int_enter at C:/esp/ESP8266_RTOS_SDK/components/freertos/port/esp8266/os_cpu_a.S:51

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 6060, room 16
tail 12
chksum 0xee
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12
tail 12
chksum 0x67
ho 0 tail 12 room 4
load 0x3ffe8420, len 3288, room 12
tail 12
chksum 0x89
csum 0x89EUUQUVZU**P+UUUժUUZUԪZ(UZUTUU*ZUXiUP*UUTUU*ZUTUUTYUTUUTUUTJUZUTU+UPIUUUTUUT*uUTZUUUUJUTUUTU*uQ(*RUZU
EUUQUUVZ
T(UP.U]UZJPUVUEUUQUVZ-RQQQQQEUTUUTQZ
ծժU)Q
QQUQQQUTUUTZURZRkZE,PQQQQQQQEQUTUUUZ+PKZTTUUTUUTUլVVUZQTQ(QTQTQTUTQUU(UE*U(JUUUPU-UVVUZTQ(QUTQTUUU\U*UUUXU
奢UUP*U-UVVUZUTQ(Qe*YTQeQU,Q)Q(UBTUEUUQUUU-UVVUZUTQ(Q(QTQQZQ(QQe*TUTUUUBXVVUZTQ(QũTQTQZQPBUU*TUTUUUBZTJPXU몱T
]UZJPQU
```

galuschka, 2021-03-25:
I get the same error. (using xtensa-lx106-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz from Espressif)
0x4025a1d5: uw_init_context_1 at /home/wjg/Repo/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/libgcc/unwind-dw2.c:1567
(discriminator 1)
What makes me wonder is the version 4.8.2 in the path shown: /home/wjg/Repo/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2
May the precompiled binaries in the tar.gz were build with mixed versions.

However: is there a working toolchain?

cassio-sdev, 2021-05-31:
Is there any update?

shuai132, 2021-07-01:
# same issue on mac:
## toolchain version
xtensa-lx106-elf-c++ (crosstool-NG esp-2020r3-49-gd5524c1) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

## backtrace
abort() was called at PC 0x402566a5 on core 0
0x402566a5: uw_init_context_1 at /home/wjg/Repo/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/libgcc/unwind-dw2.c:1567 (discriminator 1)

Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x402155ba  PS      : 0x00000030  A0      : 0x402155b8  A1      : 0x3ffeb1b0

hzarkov, 2023-01-11:
Issue is still present. I've tried using older version, but shared_ptr in older version uses 13869 IRAM(libstdc++.a) and I've end up out of IRAM. I failed in forcing libstdc++.a to be in flash so I've decided to get back to the latest version. Here using shared_ptr is not adding any libstdc++.a to the IRAM, but it's crashing... Is there a workaround/solution to the problem that I can use?

dsptech, 2023-01-12:
Hi @hzarkov,
I don't have a fix for this but one attempt might be to replace again the toolchain.
Due some comment I read, seems that the Arduino repository (https://github.com/esp8266/Arduino) use its own GCC10.7 for Esp8266.
I have no information but such a compiler and have never investigated, but if true, this could potentially solve your problem. Sadly, any further development of this RTOS framework is halted (see pinned messages) and I doubt this issue will be ever solved.
Regards.

alexCajas, 2023-04-03:
The problem still there (lastest - 35375c17), I tried to use olders version of xtensa, but a lot problems in compile time was fired. And use older versions of the SDK is not a option to. Some idea?