espressif/ESP8266_RTOS_SDK #992: Any way to remove the pthread component from the SDK v3.3?

windowsair, 2020-10-19:
## Environment

- Development Kit:
- IDF version https://github.com/espressif/ESP8266_RTOS_SDK/releases/tag/v3.3
- Development Env:      NA
- Operating System:    NA
- Power Supply:         NA

## Problem Description

It seems that `pthread` components are mandatory now. This will increase the size of the firmware. Any idea to remove it?
#840

dsptech, 2020-10-20:
Hi, this issue is already on:
https://github.com/espressif/ESP8266_RTOS_SDK/issues/849#issue-585704530

windowsair, 2020-10-21:
> Hi, this issue is already on:
> [#849 (comment)](https://github.com/espressif/ESP8266_RTOS_SDK/issues/849#issue-585704530)

oh...I'm not sure if the developer is aware of this issue...