espressif/ESP8266_RTOS_SDK #994: Using TimeZone causes a crash

ds21h, 2020-10-19:
## Environment

- Development Kit: ESP8266_RTOS_SDK Master
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 3aeaa893
- Development Env:      Make
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

After upgrading the development kit to the latest version (the former version I used displayed v3.2-526-gfe99ec40-dirty as the IDF version) the TimeZone function causes a crash.

### Expected Behavior

The TimeZone function should perform as it did: just calculate the proper local time.

### Actual Behavior

The system crashes.

### Steps to repropduce

1. Connect the system to a internet enabled AccessPoint
2. Set a SNTP server
2a. Set the TimeZone (setenv() and tzset() functions.
3. Get the current time (time() function)

### Code to reproduce this issue

Original code. This worked fine in the previous Development kit:

	sntp_setservername(0, "0.nl.pool.ntp.org");
	sntp_setservername(1, "1.nl.pool.ntp.org");
	sntp_setservername(2, "2.nl.pool.ntp.org");
	setenv("TZ", "CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00", 1);
	tzset();
	sntp_init();

Using this code in a program that displayes the current time every 10 seconds gives the following output: 

[1B][0;32mI (10513) wifi:state: 0 -> 2 (b0)
[1B][0m[1B][0;32mI (10515) wifi:state: 2 -> 3 (0)
[1B][0m[1B][0;32mI (10517) wifi:state: 3 -> 5 (10)
[1B][0m[1B][0;32mI (10527) wifi:connected with JEHok, aid = 4, channel 8, HT20, bssid = 9c:80:df:28:1e:ef
[1B][0mStation Connected
Station got IP '192.168.2.89'
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core 0 register dump:
PC      : 0x402244e1  PS      : 0x00000030  A0      : 0x4022449b  A1      : 0x3ffecf30  

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3ffea4a4  

A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000031  A9      : 0x000000b0  

A10     : 0x402699c2  A11     : 0x0000000a  A12     : 0x4026aafc  A13     : 0x3ffea37c  

A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  

Backtrace: 0x402244e1:0x3ffecf30 0x40224444:0x3ffecf70 0x40222030:0x3ffecf80 0x40222135:0x3ffecf90 0x40255be9:0x3ffecfa0 0x40236998:0x3ffecfc0 0x40236a67:0x3ffed000 

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 7888, room 16 
tail 0
chksum 0x34
load 0x3ffea408, len 48, room 8 
tail 8
chksum 0x34
load 0x3ffea438, len 3760, room 0 
tail 0
chksum 0x6e
csum 0x6eQØ“Í	[08]pL@É^uÀB¾M	5áÄ!ìB[11]%I![0C]P[11]ñJiNPãT7Ò(ÞTÞÔÕyÄ)Þ[1F]JC‚[19]	%[[0E][01]òTPQˆÒYH[11]ƒ-[13][1F]‰NX¢[17]ñ
	m[02]1mÏ[È[07])A‰AËPK^qˆÔRÐÊZÐÖ+¢OCÞ[1F]JCÂ[11]	%Y[0E][01][03]@5Ô…[08][01][01][1D][11]+kYÐNx¢[17]ñ
	Š	%[[0C][01][03]@)Q
[01][01][01][1D]3mÊê
YÐ›ÅJ	][0B]%[0C][01][03]
%C
CMi
[15][11])P[05]+[00]E “ÅJ	][0B]%[0C][01]æ£ÏÊ[€[08]ÊGÔ)!EØ‘Í	[08]xL@Éy[0C][01]š€CÉ|[01][01][01][01][01][01]®È[08][01][01][01][01][01]XT7 QKVP[01]SËTÊ‰OCÞ[1F][0F][00]zFBÉ[[0C][01][11]-Z[01]![01][01][01][01][01]¬IPLP[01][01][01][11]
[11][04][11][19][11][01][11][11][11][11]QØ[19]AÔ[1F]ƒJA[02][13][0B]%^u[00][13]1ïoÍ[00][01][01][01][01]#PNP[01]![01][01][11]
[11]
[11][11]%[11][01][11][11]Š[11][1F]¢MCÔ[1D]JC@[15][01]%Y[0E]![15]%.Üa,[01][01][01][01][01]¢M@[0C][00][01][01][11][01][11][11][11]‚[11][11][11][11]E[0C][11]QØNX‰›ÅJ	Š[11][0B]%[0C][01]l[08]-[07]LZPÐY€PâÖQØ[19]AÔ[1F]‹J	Š[11][0B]#ƒö[0F][07]5lÊ‹[00][15]1DTó[04]X[11]Š[11]Š[01]R…Í[11]T[15]Š[11]Š[01]ÓÏ‹9ê[11]Œ[01]j„[19][00])„ÊƒÊ[00]GD[15]ñ[08][01]‘F@JØôIËX7VT[13]‹Lt×‹9[11][15]M	[01]‡ˆÍ[11]\[15][1D]bX5¯Ö[11]Z[11]¢[00][01]M]E[01]ŽÞ+)EØ‘Í	[08]`[0E][0B]#‹ÎÞÖu@nO‹[00][1D]1L[08][06]9[11][1D][15]¢€P…Í[11]Z•
[11][11]7¯Ö[11]P[15]Š[01][01][13]UDBÊCHÔ)!EØ‘Å[01][08]`N[0B]#ƒÎÞÖu@Ö­n[0C][17][01]N[08][06]9[11][15]e…[01]P…Í[11]\[13][11][11][01]ÓØm[04]P[11][11]	[01][13]Á[0B])ˆGÄ[13]CÔ[15]JA[00][1D][02][01]ÝÌôIËX7VT[11][01]N[08]‹9[11][15]eH€ˆTó[04]\[11][11][19][01]ÓØm[0C]Ðm†[01][01]‹‹[15][0B])NÞ))GD[15]ñ
[01]ÁD@ÉY[0E][01]YˆV[05][08]P%û^1®ôL[0B]#[07]‰Í[0C][11][04][11][11]QØ[11]á

Commenting out the two lines setting the TimeZone gives the following output:

[1B][0;32mI (10514) wifi:state: 0 -> 2 (b0)
[1B][0m[1B][0;32mI (10518) wifi:state: 2 -> 3 (0)
[1B][0m[1B][0;32mI (10522) wifi:state: 3 -> 5 (10)
[1B][0m[1B][0;32mI (10532) wifi:connected with JEHok, aid = 4, channel 8, HT20, bssid = 9c:80:df:28:1e:ef
[1B][0mStation Connected
Station got IP '192.168.2.89'
[1B][0;32mI (11304) tcpip_adapter: sta ip: 192.168.2.89, mask: 255.255.255.0, gw: 192.168.2.254[1B][0m
Counting....20, Time 2020-10-19 13:59:06
Counting....30, Time 2020-10-19 13:59:16

r126401, 2020-10-22:
I have the same problem

ds21h, 2020-10-26:
The problem is situated in the setenv() call.
After including the setenv.c in the project the problem is solved.
As the setenv() function is part of stdlib I used a version I found on internet. In order to have it compiled I needed to add the following includes/defines:
#include "string.h"
#include "stdlib.h"
# define __glibc_unlikely(cond)	__builtin_expect ((cond), 0)
# define __glibc_likely(cond)	__builtin_expect ((cond), 1)

Please update the SDK/Toolchain as the version provided now clearly has an error.

gusensheng, 2020-10-28:
unsubscribe
<https://github.com/

thanks.

colesnicov, 2020-11-08:
I have this problem too.

colesnicov, 2020-11-13:
- this is not solved.  I do not have a setenv.c file on my computer.  
```
locate setenv.c
```
 did not find anything.  Espressif, why not answer?  It is not possible to set the time zone.  it tells me that the error is in **SETENV_R(...)** , i think, and when I get to the computer, **I will add the output from the console!**

ds21h, 2020-11-13:
The setenv() is included in stdlib. This is provided as a library in the toolchain (so allready compiled for the ESP8266). stdlib is an open source development so it is freely available in the internet. I used a search in Google for "c setenv source". It produces a large number of results. I used "https://code.woboq.org/userspace/glibc/stdlib/setenv.c.html" (the first result).
Make a copy of that setenv.c file in your own project. As setenv() is then present in your project the linker will not look for it so that version is used and not the one provided in the toolchain.
In order to have it compiled you need to add the includes and defines I described earlier.

This is still a workaround. I am also waiting for a working version of the toolchain.

Hope this helps

colesnicov, 2020-11-13:
I'm sorry. I got it a little confused. Of course I have the setenv.c file in the toolchain. Here is a statement from the console:

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40225ec1  PS      : 0x00000030  A0      : 0x40225e7b  A1      : 0x3ffef7b0  
0x40225ec1: _setenv_r at /builds/idf/crosstool-NG/.build/xtensa-lx106-elf/src/newlib/newlib/libc/stdlib/setenv_r.c:89 (discriminator 1)

0x40225e7b: _setenv_r at /builds/idf/crosstool-NG/.build/xtensa-lx106-elf/src/newlib/newlib/libc/stdlib/setenv_r.c:70

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3ffea98c  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000090  
A10     : 0x0000002a  A11     : 0x00000029  A12     : 0x402909ec  A13     : 0x3ffea864  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  

Backtrace: 0x40225ec1:0x3ffef7b0 0x40225e24:0x3ffef7f0 0x40222a95:0x3ffef800 
0x40225ec1: _setenv_r at /builds/idf/crosstool-NG/.build/xtensa-lx106-elf/src/newlib/newlib/libc/stdlib/setenv_r.c:89 (discriminator 1)

0x40225e24: setenv at /builds/idf/crosstool-NG/.build/xtensa-lx106-elf/src/newlib/newlib/libc/stdlib/setenv.c:40

0x40222a95: task_sntp at task_sntp.c:79
```

ds21h, 2020-11-13:
I don't have the setenv.c file in my toolchain. I only have object code in a library.
Have you tried to include a setenv.c file from the internet in your project?

colesnicov, 2020-11-13:
Yes. I can to tray this.

colesnicov, 2020-11-13:
Yes it works. But the time zone is wrong. is shifted by one hour. I set the time zone for CTS-2 but I get the time for zone CST-3!