espressif/ESP8266_RTOS_SDK #995: printf() and sprintf() not working for doubles

ds21h, 2020-10-22:
## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 3aeaa893
- Development Env:      Make
- Operating System:    Windows
- Power Supply:         USB

## Problem Description

The formatting of doubles using the printf() or sprintf() function doesn't work. As the result thereoff the cJSON library included in the SDK doesn't function either for creating JSON with numeric content as there the sprintf() function is used.
The previous SDK/Toolchain functioned just fine.

### Expected Behavior

Functionality according to the specifications.

### Actual Behavior

No output is produced.

### Steps to reproduce

Every call to printf() or sprintf() that tries to format a double creates no output.

### Code to reproduce this issue

Execution of the following code:
	double lValue = 2.4;

	printf("Value as Integer: %d\n", (int)lValue);
	printf("Value as double using e function: %1.5e\n", lValue);
	printf("Value as double using f function: %1.5f\n", lValue);
	printf("Value as double using g function: %1.5g\n", lValue);
	printf("End of test\n");

Results in the following output:
Value as Integer: 2
Value as double using e function: 
Value as double using f function: 
Value as double using g function: 
End of test

zinahe, 2020-10-24:
Hi, 

For floating point support in functions like **printf() and scanf()**, change the **newlib** config in your project **sdkconfig** as follows:

```
CONFIG_NEWLIB_LIBRARY_LEVEL_NORMAL=y
CONFIG_NEWLIB_LIBRATY_LEVEL_NANO=
```

Hope that helps.

ds21h, 2020-10-24:
Hi Zinahe,
Thanks for the tip.
Those config labels are not present in my SDKCONFIG. However 'un'setting CONFIG_NEWLIB_NANO_FORMAT also did the trick.
Is this behaviour documented somewhere?
What are nano formatting options?

closed 2020-10-26 by ds21h

dsptech, 2020-12-11:
Just for the record, after the upgrade to GCC8.4 this option also affect the print capability of uint8_t. In "lib nano"  mode, the output of:
`printf("Number8: %" PRIu8 "\n",(uint8_t) 1);`
 is: "Number8: hu". 
Regards.