espressif/ESP8266_RTOS_SDK #996: Allow WiFi sniffer to dump frames with wrong FCS

doragasu, 2020-10-27:
## Environment

- Development Kit: nodemcu (ESP-12E)
- IDF version release/v3.3, commit: `210943a9`
- Development Env: other (vim)
- Operating System: Ubuntu
- Power Supply: USB/battery

## Problem Description

I have been testing the WiFi Sniffer using the example code. I got it to start and I can see frames with correct FCS (Frame Check Sequence), but I am not able to see frames with wrong FCS. I need to be able to monitor packets with incorrect FCS for my application. This can be achieved on a Linux PC with compatible drivers with command: `# iw dev <radio_device> set monitor fcsfail`. But it seems there is no way to do this on ESP8266/ESP32

### Expected Behavior

WiFi Sniffer allows to dump packets with wrong FCS. This could be disabled by default, but enabled with an API call, like e.g.: `esp_wifi_set_fcsfail(boolean enable)`

### Actual Behavior

WiFi Sniffer does not show packets with wrong FCS, only packets with valid FCS field are dumped. It seems there is no API call to enable these packets to be dumped.

### Steps to repropduce

1. Build, flash and start (monitor) the `sniffer` example.
2. On a PC, generate WiFi frames with invalid FCS. Make sure they are generated on the same WiFi channel the sniffer is configured in.
3. Browse the frames captured by the sniffer. The frames with incorrect FCS will not be shown.

dawidchyrzynski, 2021-02-11:
@doragasu Have you found any workaround for this? I need to count packets with invalid FCS but it doesn't seem to be possible.

doragasu, 2021-02-11:
Hi,

I did not find a workaround using an ESP8266, but this was finally implemented for ESP32 chips. Setting up the filter might be a bit tricky, but works. You can see more info in this issue: https://github.com/espressif/esp-idf/issues/6473