espressif/ESP8266_RTOS_SDK #1001: PWM broken since commit 1366f6dd

MitchBradley, 2020-11-05:
The PWM example in examples/peripherals/pwm does not work in the latest master, when compiled using the default sdkconfig.  None of the GPIO pins toggle.  Some testing shows that the interrupt handler is not being called.

I bisected the commit history and found the problem.  Commit 5a916cd7 works, but the next version - the  merge to 1366f6dd - does not work.

The problem appears to be the removal of these lines from components/esp8266/source/startup.c:
```
    /*enable tsf0 interrupt for pwm*/
    REG_WRITE(PERIPHS_DPORT_BASEADDR, (REG_READ(PERIPHS_DPORT_BASEADDR) & ~0x1F) | 0x1);
    REG_WRITE(INT_ENA_WDEV, REG_READ(INT_ENA_WDEV) | WDEV_TSF0_REACH_INT);

```
If those lines are reinstated, PWM works as expected.  I tried adding those line back to startup.c - that worked.  Then I moved them to pwm_example_main.c - that worked too.  Finally, I moved them to the pwm_init() function in pwm.c, and that also worked.

I suspect that this might also affect the NMI WDT, but I have not tested that.

MitchBradley, 2020-11-06:
Only the first line - for PERIPHS_DPORT_BASEADDR - is necessary to fix the problem.  The INT_ENA_WDEV line is unnecessary because that register is already set correctly, probably by code elsewhere.

colesnicov, 2020-11-08:
this is not work for me :/