espressif/ESP8266_RTOS_SDK #1015: 烧录esp9200的地址选择

woshi520laoxie, 2020-11-24:
1、其实想解决的问题：
      esp8266目前的AT指令是否能够使用http指令

1.1、今天我想试试esp8266如果更新到最新的固件能否使用http协议，但是一直没有人回复我
所以我到官网下载了，一个比较新的固件，想试试看，ESP8266-IDF-AT_V2.0
这个固件有一个文档，我按照文档修改了，写入的地址，但是模组无法运行。
1.2、所以我提出了第二个问题，如果下载这个 ESP8266-IDF-AT_V2.0的固件

--flash_mode dio 
--flash_freq 40m 
--flash_size 2MB 

0x0000 bootloader/bootloader.bin 
0x8000 partitions_at.bin
0x9000 ota_data_initial.bin 
0x10000 esp-at.bin 
0xBF000 at_customize.bin 
0xC0000 customized_partitions/factory_param.bin 
0xC1000 customized_partitions/server_cert.bin 
0xC3000 customized_partitions/server_key.bin 
0xC5000 customized_partitions/server_ca.bin 
0xC7000 customized_partitions/client_cert.bin 
0xC9000 customized_partitions/client_key.bin 
0xCB000 customized_partitions/client_ca.bin 

并不太知道这个问题出在哪里，请问有朋友知道的吗？
[image]

zheyuejiang, 2020-12-23:
@woshi520laoxie 没有用过这个AT，但是图片上的crystal是26M，你现在是40M，修改一下，不知道行不行，你可以试试

woshi520laoxie, 2020-12-23:
@zheyuejiang 好的，谢谢