espressif/ESP8266_RTOS_SDK #1016: ESP8266 Toolchain for Raspberry PI

jumjum123, 2020-11-24:
There is no toolchain available for ESP8266, which is compiled for Raspberry Pi and/or other ARM servers.
ESP32 supports this (https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-linux-armel.tar.gz), it would be great to get it for ESP8266 similiar way.

Just seen first info for RISC based follower for ESP8266. Would be great to have an ARM based toolchain from beginning.;-)

digits122, 2021-04-01:
try this
apt-get install gcc-xtensa-lx106

danlkv, 2023-12-20:
Installing with `apt` did not work for me since it did not include the library files. Compilation failed on including `string.h`. 

It did work however, downloading the archive for `armel`.
```
wget https://dl.espressif.com/dl/xtensa-lx106-elf-gcc8_4_0-esp-2020r3-linux-armel.tar.gz
```
The toolchain worked with hello world example.

Perhaps an update in README.md would help people choose the right arch.