espressif/ESP8266_RTOS_SDK #1017: Change Monitor baud rate issue

Gustbel, 2020-11-24:
## Environment

- Development Kit:      Any ESP8266EX
- IDF version:     89c81f7
- Development Env:     make - idf.py
- Operating System:     Ubuntu
- Power Supply:         [USB|external 5V

## Problem Description

When I change the configuration of *"menuconfig->Serial flasher config ->'make monitor'"* baud rate to another baud rate, the baud rate does not change and remains in its default configuration 78840 bps. I was only able to fix it by modifying the sdkconfig file replacing the values where it found 78840.

### Steps to repropduce

The problem arises without doing any special configuration.

Thanks!

zheyuejiang, 2020-12-22:
[image]

zheyuejiang, 2020-12-22:
try this