espressif/ESP8266_RTOS_SDK #1019: Unable to build using mbedtls component ( undefined reference to `mbedtls_chachapoly_init')

zinahe, 2020-12-16:
## Environment

- Development Kit: Wemos mini-D1
- IDF version: 1be2289f
- Development Env:      [Make]
- Operating System:     [Ubuntu]
- Power Supply:         [USB]

## Problem Description

I wanted to experiment with using mbedtls encryption functions, in particular the ones located in  ```chachapoly.h```. However even the simplest example shown below compiles fine but fails during linking because it is missing the mbedtls object binaries.

### Code to reproduce this issue
```c
#include "esp_system.h"
#include "mbedtls/chachapoly.h"

void app_main() {
    mbedtls_chachapoly_context ctx;
	
    mbedtls_chachapoly_init( &ctx );

    mbedtls_chachapoly_free( &ctx );
	
    while(true);
}
```

## make 
```
build/main/libmain.a(user_main.o):(.literal.app_main+0x0): undefined reference to `mbedtls_chachapoly_init'
build/main/libmain.a(user_main.o):(.literal.app_main+0x4): undefined reference to `mbedtls_chachapoly_free'

collect2: error: ld returned 1 exit status

```

Any help is greatly appreciated.

Thanks

ln-12, 2021-07-22:
@zinahe Hey, did you find a solution for that problem?

zinahe, 2021-07-24:
@ln-12,

Sorry no. I don't remember what happened. I moved on. As you can see, the developers never bothered to respond. I think the esp8266 platform is doomed at this point. No point wasting your time on it anymore.

closed 2021-07-24 by zinahe