espressif/ESP8266_RTOS_SDK #1020: 关于环境搭建的异常

woshi520laoxie, 2020-12-18:
大家好:
       今天我在搭建esp8266开发环境的时候出现如下问题：
ESP8266_RTOS_SDK  下面解压了压缩文件，esp-idf-master.zip
进入到esp-idf
1、执行指令
./install.sh
报如下错误
[image]

请问这个问题是如何处理的

woshi520laoxie, 2020-12-18:
根据ReadMe中介绍的
配置IDF_PATH
1、export IDF_PATH=/home/sambaShare/esp/esp8266/esp8266_sdk_3.3
配置交叉编译器路径
2、export ESP8266TOOLCHAIN=/home/sambaShare/esp/esp8266/toolchain_xtensa/xtensa-lx106-elf/bin
进入到hello_world
3、cd /home/sambaShare/esp/esp8266/esp8266_sdk_3.3/examples/get-started/hello_world
4、make menuconfig
这些都是对的，但是
执行make  报错
[image]

woshi520laoxie, 2020-12-18:
我切换了python版本为使用python3之后
1、执行./install.sh
[image]

2、. ./export.sh
[image]

woshi520laoxie, 2020-12-20:
安装教程介绍的
1、添加交叉编译器 xtensa-lx106-elf-gcc -v
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-100-ge567ec7)

添加 IDF_PATH
export IDF_PATH=/home/sambaShare/esp/esp8266/esp8266_sdk_3.3
export PATH=$PATH:$IDF_PATH

2、make menuconifg

3、make报错
WARNING: Missing submodule components/mbedtls/mbedtls...
Attempting 'git submodule update --init components/mbedtls/mbedtls' in esp-idf root directory...
20:42:58.724442 git.c:576               trace: exec: git-submodule update --init components/mbedtls/mbedtls
20:42:58.724511 run-command.c:646       trace: run_command: git-submodule update --init components/mbedtls/mbedtls
20:42:58.733840 git.c:344               trace: built-in: git rev-parse --git-dir
20:42:58.735925 git.c:344               trace: built-in: git rev-parse --git-path objects
20:42:58.737771 git.c:344               trace: built-in: git rev-parse -q --git-dir
20:42:58.740404 git.c:344               trace: built-in: git rev-parse --show-prefix
20:42:58.742564 git.c:344               trace: built-in: git rev-parse --show-toplevel
20:42:58.744985 git.c:344               trace: built-in: git submodule--helper init components/mbedtls/mbedtls
error: pathspec 'components/mbedtls/mbedtls' did not match any file(s) known to git.
/home/sambaShare/esp/esp8266/esp8266_sdk_3.3/make/project.mk:630: recipe for target '/home/sambaShare/esp/esp8266/esp8266_sdk_3.3/components/mbedtls/mbedtls/.git' failed
make: *** [/home/sambaShare/esp/esp8266/esp8266_sdk_3.3/components/mbedtls/mbedtls/.git] Error 1

这个问题，请问有朋友知道的吗

woshi520laoxie, 2020-12-20:
朋友好：
      我用国内的镜像进行git之后，编译有进展了，但是还有一个错误 cannot find -lc_nano

cannot find -lc_nano

/home/sambaShare/esp/esp8266/toolchain_xtensa/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.2.0/../../../../xtensa-lx106-elf/bin/ld: cannot find -lc_nano
collect2: error: ld returned 1 exit status
/home/sambaShare/esp/esp8266/xwz_esp8266_gitee/ESP8266_RTOS_SDK/make/project.mk:509: recipe for target '/home/sambaShare/esp/esp8266/xwz_esp8266_gitee/ESP8266_RTOS_SDK/testHello/hello_world/build/hello-world.elf' failed
make: *** [/home/sambaShare/esp/esp8266/xwz_esp8266_gitee/ESP8266_RTOS_SDK/testHello/hello_world/build/hello-world.elf] Error 1

zheyuejiang, 2020-12-21:
你这个第一张图是python依赖没有对吧

zheyuejiang, 2020-12-21:
而且你是要编译esp8266还是esp32，要先弄清楚

woshi520laoxie, 2020-12-21:
@zheyuejiang 我编译的是esp8266

woshi520laoxie, 2020-12-21:
@zheyuejiang 一开始是遇到了一些这些编译问题，现在编译esp_idf 是可以正常的了，应该是不需要编译esp_idf的吧，我看到官网提供的编译流程，目前编译 8266的例子hello world的时候报 cannot find -lc_nano

zheyuejiang, 2020-12-21:
你把你的工程的目录贴一下，还有环境变量放上来看看

woshi520laoxie, 2020-12-22:
@zheyuejiang 你好啊，之前搭建由于操作有误导致的，目前已经可以编译了，但是我编译的时候一直提示我输入gitee的账号密码

Username for 'https://gitee.com': 563376881@qq.com
Password for 'https://563376881@qq.com@gitee.com': 

请问这个账号密码应该是什么呢？输入我的gitee账号密码也一直提示不对。