espressif/ESP8266_RTOS_SDK #1021: cannot find -lc_nano

woshi520laoxie, 2020-12-20:
大家好：
      我在搭建esp8266环境的时候遇到了一个错误  cannot find -lc_nano
      但是这个异常，百度没有找到问题的解决。
有朋友知道的吗？

下面贴出部分异常提示
/home/sambaShare/esp/esp8266/toolchain_xtensa/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/5.2.0/../../../../xtensa-lx106-elf/bin/ld: cannot find -lc_nano
collect2: error: ld returned 1 exit status
/home/sambaShare/esp/esp8266/xwz_esp8266_gitee/ESP8266_RTOS_SDK/make/project.mk:509: recipe for target '/home/sambaShare/esp/esp8266/xwz_esp8266_gitee/ESP8266_RTOS_SDK/testHello/hello_world/build/hello-world.elf' failed
make: *** [/home/sambaShare/esp/esp8266/xwz_esp8266_gitee/ESP8266_RTOS_SDK/testHello/hello_world/build/hello-world.elf] Error 1

woshi520laoxie, 2020-12-21:
这个问题的出现是在，编译esp8266 的 hello world的时候出现的异常

carolcabral, 2021-02-21:
Same problem over here. Any solution?

carolcabral, 2021-02-22:
Got it to work by changing a configuration in make menuconfig:
https://www.esp8266.com/viewtopic.php?p=89409

studyNowHa, 2022-07-22:
15:28:04 **** Incremental Build of configuration Default for project ESP8266_RTOS_SDK ****
make all 
**Toolchain path: /opt/arm-nono-eabi-gcc/bin/xtensa-lx106-elf-gcc**
**WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-92-g8facf4c0
**Expected to see version: esp-2020r3-49-gd5524c1**
**Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.**
**WARNING: Compiler version is not supported: 5.2.0****
Expected to see version(s): 8.4.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from D:/espressif/ESP8266/ESP8266_RTOS_SDK\requirements.txt are satisfied.
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-92-g8facf4c0
Expected to see version: esp-2020r3-49-gd5524c1
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 5.2.0
Expected to see version(s): 8.4.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
App "hello-world" version: v3.4-63-ga192988b-dirty
LD /d/espressif/ESP8266/ESP8266_RTOS_SDK/examples/get-started/hello_world/build/hello-world.elf
d:/espressif/ai-thinker/aithinkeride_v1.0/msys32/opt/arm-nono-eabi-gcc/bin/../lib/gcc/xtensa-lx106-elf/5.2.0/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lc_nano
collect2.exe: error: ld returned 1 exit status
make: *** [D:\espressif\ESP8266\ESP8266_RTOS_SDK\/make/project.mk:510: /d/espressif/ESP8266/ESP8266_RTOS_SDK/examples/get-started/hello_world/build/hello-world.elf] Error 1

15:35:14 **** Incremental Build of configuration Default for project ESP8266_RTOS_SDK ****
make all 
**Toolchain path: /opt/arm-nono-eabi-gcc/bin/xtensa-lx106-elf-gcc
Toolchain version: esp-2020r3-49-gd5524c1
Compiler version: 8.4.0**
Python requirements from D:/espressif/ESP8266/ESP8266_RTOS_SDK\requirements.txt are satisfied.
App "hello-world" version: v3.4-63-ga192988b-dirty
LD /d/espressif/ESP8266/ESP8266_RTOS_SDK/examples/get-started/hello_world/build/hello-world.elf
esptool.py v2.4.0
To flash all build output, run 'make flash' or:
python /d/espressif/ESP8266/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py --chip esp8266 --port COM1 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size 2MB 0x0 /d/espressif/ESP8266/ESP8266_RTOS_SDK/examples/get-started/hello_world/build/bootloader/bootloader.bin 0x10000 /d/espressif/ESP8266/ESP8266_RTOS_SDK/examples/get-started/hello_world/build/hello-world.bin 0x8000 /d/espressif/ESP8266/ESP8266_RTOS_SDK/examples/get-started/hello_world/build/partitions_singleapp.bin

15:35:30 Build Finished (took 16s.256ms)