espressif/ESP8266_RTOS_SDK #1022: os_delay_us() 函数不能准确定时，偏差非常严重（10倍以上）

rongshengqin, 2020-12-21:
因为写i2c需要用到延时函数，按照https://sub.nanona.fi/esp8266/timing-and-ticks.html一文中方法，测试了一下 os_delay_us() 函数的精确度。方法是读取CCOUNT 寄存器的值，这个值代表cpu的 tick数。由于cpu频率是80MHZ，所以每个tick的时间是1/80000000 s，也就是12.5 ns。用函数读取前后tick数的差值就可以算出函数执行的时间。
    读取tick数的函数为：
static inline int32_t asm_ccount(void) {
    int32_t r;
    asm volatile ("rsr %0, ccount" : "=r"(r));
    return r;
}
测试函数为
static uint32_t __measure(void) {
    int32_t btime, etime;
    btime = asm_ccount();
    os_delay_us(1); //延时一us
    etime = asm_ccount();
    return ((uint32_t)(etime - btime));
}
   理论上__measure()返回值应该是 1us/12.5ns=80，加上函数overhead应该在一百左右。但在我的板子上运行却返回1447个tick，是理论值十多倍。值得注意的是，当在esp-open-rtos下编译运行（os_delay_us替换成sdk_os_delay_us）的时候却返回了107，非常接近理论估算。
     希望有人能同样测试一下os_delay_us()。另外sdk_os_delay_us和os_delay_us应该是同一个函数把？

zheyuejiang, 2020-12-22:
我测试了一下，os_delay_us(int us)，us>=5的时候是正常的，每us是84个tick左右

zheyuejiang, 2020-12-22:
你试一下这个代码，我测出来os_delay_us(1)，大概在116

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"

static inline int32_t asm_ccount(void) {
    int32_t r;
    asm volatile ("rsr %0, ccount" : "=r"(r));
    return r;
}

static uint32_t __measure(void) {
    os_delay_us(10000); //延时一us
    int32_t btime, etime;
    btime = asm_ccount();
    os_delay_us(1); //延时一us
    etime = asm_ccount();
    return ((uint32_t)(etime - btime));
}

void app_main()
{
    printf("Hello world!\n");
    printf("******** %d\n", __measure());
    /* Print chip information */
    esp_chip_info_t chip_info;
    esp_chip_info(&chip_info);
    printf("This is ESP8266 chip with %d CPU cores, WiFi, ",
            chip_info.cores);

    printf("silicon revision %d, ", chip_info.revision);

    printf("%dMB %s flash\n", spi_flash_get_chip_size() / (1024 * 1024),
            (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");

    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
}

rongshengqin, 2020-12-22:
@zheyuejiang  我跑了一下你给的代码，返回221，偏差还是很大；如果把os_delay_us(10000)这句注释掉，返回2301。不知道怎么回事，我得到的结果就是不正常，非常奇怪。

zheyuejiang, 2020-12-22:
你的RTOS的版本是多少，我的是3.3

rongshengqin, 2020-12-22:
@zheyuejiang  我也是3.3。不纠结了 ，反正这个函数实现有问题，不用就行了。感觉这个sdk很多BUG。

zheyuejiang, 2020-12-23:
@rongshengqin ,是的，没有错，还有很多坑，哈哈哈哈哈

dsptech, 2020-12-30:
Hi @rongshengqin,
Can this comment and related issue be helpful to you?

https://github.com/espressif/ESP8266_RTOS_SDK/issues/680#issuecomment-522870764