espressif/ESP8266_RTOS_SDK #1023: Infrared Transmission not working properly (stuck indefinintely on semaphore)

Ordinateur-Hack, 2020-12-21:
## Environment

- Development Kit: ESP8266 Node MCU board
- IDF version: commit 7f99618d
- Development Env:      VSCode
- Operating System:    Windows
- Power Supply:         USB

## Problem Description
I'm trying to get the [`peripherals/ir_rx example`](https://github.com/espressif/ESP8266_RTOS_SDK/blob/7f99618d9e27a726a512e22ebe81ccbd474cc530/examples/peripherals/ir_tx/main/ir_tx_example_main.c) to run, so that I can send IR signals. Receiving IR signals is working fine.

### Expected Behavior
When I run the program, I should be able to see [this line](https://github.com/espressif/ESP8266_RTOS_SDK/blob/7f99618d9e27a726a512e22ebe81ccbd474cc530/examples/peripherals/ir_tx/main/ir_tx_example_main.c#L47) in the monitor output:

```cpp
ESP_LOGI(TAG, "ir tx nec: addr:%02xh;cmd:%02xh;repeat:%d", ir_data[0].addr1, ir_data[0].cmd1, 4);
```

### Actual Behavior
The program is stuck [on line 46](https://github.com/espressif/ESP8266_RTOS_SDK/blob/7f99618d9e27a726a512e22ebe81ccbd474cc530/examples/peripherals/ir_tx/main/ir_tx_example_main.c#L46) and no IR signal is sent at all.

After several hours of debugging, I found out that the program is stuck [here on line 277](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/esp8266/driver/ir_tx.c#L277), where it can't take the semaphore. The example for IR transmission specifies the timeout to be equal to `portMAX_DELAY`, which according to [FreeRTOS](https://www.freertos.org/a00122.html) means:

> If INCLUDE_vTaskSuspend is set to ‘1’ then specifying the block time as portMAX_DELAY will cause the task to block indefinitely (without a timeout).

This is exactly what I am experiencing. Nothing is happening after reaching [line 277](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/esp8266/driver/ir_tx.c#L277) and thus there is no log output [here](https://github.com/espressif/ESP8266_RTOS_SDK/blob/7f99618d9e27a726a512e22ebe81ccbd474cc530/examples/peripherals/ir_tx/main/ir_tx_example_main.c#L47).

I've also tried to change the ux priority [here](https://github.com/espressif/ESP8266_RTOS_SDK/blob/7f99618d9e27a726a512e22ebe81ccbd474cc530/examples/peripherals/ir_tx/main/ir_tx_example_main.c#L58) to 14, but without any difference in the outcome.

### Steps to repropduce
Compile [this example](https://github.com/espressif/ESP8266_RTOS_SDK/tree/7f99618d9e27a726a512e22ebe81ccbd474cc530/examples/peripherals/ir_tx).

## Debug Logs
```
gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
```
After this line, nothing happens anymore.
Note that I've also tried to combine the transmission and receiving infrared examples. I am able to receive IR signals all the time and they are printed correctly to the monitor, just the transmission seems to not work at all.

## Other items if possible
- [X] [sdkconfig file](https://gist.github.com/Ordinateur-Hack/ee54f369575caeb55b0033f46646f9c3)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

rodrigo2711, 2021-02-22:
I have the same problem.

benjaminaigner, 2021-04-19:
One of my students had the same problem.

Our solutions was a different initialisation:

```
 .timer = IR_TX_WDEV_TIMER
```

change it to 

```
 .timer = IR_TX_HW_TIMER
```
I hope you can fix this issue,
greetings

Andrey-m, 2021-08-29:
IR_RX example stuck on:
ESP_LOGI(TAG, "addr1: 0x%x, addr2: 0x%x, cmd1: 0x%x, cmd2: 0x%x", ir_data.addr1, ir_data.addr2, ir_data.cmd1, ir_data.cmd2);