espressif/ESP8266_RTOS_SDK #1044: esp-aliyun IOT_MQTT_Construct  mbedtls_ssl_fetch_input error=-0x7280

sschors, 2021-01-29:
## Environment

- Development Kit: 
      esp8285;

- IDF version: 
      esp-aliyun: commit 125902901f617e395414720f07d5eb7b901302df
      ESP8266_RTOS_SDK: commit 8189b9a4b6ba066c4f29544d1b755b5fd2426b62
      
- Development Env:
      Toolchain version: crosstool-ng-1.22.0-100-ge567ec7. From :https://dl.espressif.com/dl/xtensa-lx106-elf-linux64-1.22.0-100-ge567ec7-5.2.0.tar.gz

- Operating System:     CentOS Linux release 7.7.1908 (Core)

## Problem Description
   My project was transplanted from esp-aliyun\examples\mqtt\mqtt_example\components\mqtt_handle\mqtt_solo.c .  When MQTT started to create the connection(IOT_MQTT_Construct ), I got the following log information:

iotx_mc_init(230): MQTT init success!
wrapper_mqtt_connect(2697): calling TCP or TLS connect HAL for [1/3] iteration
Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280
Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280
Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280
...

**error=-0x7280** Printed for A few seconds before I cut off the power, And it didn't appear every time I connected. It came up a few times early on in my development, and when I tried to reproduce it after I finished the rest of the project, the problem disappeared, as if it could not be repeated.

Here are some of the logs I kept at that time:

    [19:59:42.440]ÊÕ¡û¡ô
    ets Jan  8 2013,rst cause:1, boot mode:(3,2)

    load 0x40100000, len 7160, room 16 
    tail 8
    chksum 0x0f
    load 0x3ffe8408, len 24, room 0 
    tail 8
    chksum 0x86
    load 0x3ffe8420, len 3536, room 0 
    tail 0
    chksum 0xcd
    csum 0xcd[0;32mI (45) boot: ESP-IDF v3.3-dirty 2nd stage bootloader[0m
    [0;32mI (45) boot: compile time 18:02:26[0m
    [0;32mI (45) boot: SPI Speed      : 40MHz[0m
    [0;32mI (50) boot: SPI Mode       : DOUT[0m
    [0;32mI (57) boot: SPI Flash Size : 2MB[0m
    [0;32mI (63) boot: Partition Table:[0m
    [0;32mI (68) boot: ## Label            Usage          Type ST Offset   Length[0m
    [0;32mI (80) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
    [0;32mI (91) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
    [0;32mI (103) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
    [0;32mI (114) boot:  3 ota_0            OTA app          00 10 00010000 000f0000[0m
    [0;32mI (126) boot:  4 fctry            WiFi data        01 02 00100000 00004000[0m
    [0;32mI (137) boot:  5 ota_1            OTA app          00 11 00110000 000f0000[0m
    [0;32mI (149) boot: End of partition table[0m
    [0;32mI (156) boot: No factory image, trying OTA 0[0m
    [0;32mI (163) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x887bc (559036) map[0m

    [19:59:42.862]ÊÕ¡û¡ô[0;32mI (422) esp_image: segment 1: paddr=0x000987d4 vaddr=0x402987cc size=0x14a6c ( 84588) map[0m
    [0;32mI (459) esp_image: segment 2: paddr=0x000ad248 vaddr=0x3ffe8000 size=0x0087c (  2172) load[0m
    [0;32mI (461) esp_image: segment 3: paddr=0x000adacc vaddr=0x40100000 size=0x00b80 (  2944) load[0m
    [0;32mI (471) esp_image: segment 4: paddr=0x000ae654 vaddr=0x40100b80 size=0x05b34 ( 23348) load[0m
    [0;32mI (494) boot: Loaded app from partition at offset 0x10000[0m

    [19:59:42.987]ÊÕ¡û¡ô[0;32mI (520) system_api: Base MAC address is not set, read default base MAC address from EFUSE[0m
    [0;32mI (529) system_api: Base MAC address is not set, read default base MAC address from EFUSE[0m

    [19:59:43.090]ÊÕ¡û¡ôphy_version: 1159.0, 85b471e, Apr 21 2020, 17:03:08, RTOS new
    [0;32mI (612) phy_init: phy ver: 1159_0[0m
    [0;32mI (617) reset_reason: RTC reset 1 wakeup 0 store 0, reason is 1[0m

    [0;32mI (839) ctlrConfig: SYSTEM_EVENT_STA_START[0m

    [19:59:44.887]ÊÕ¡û¡ô[0;32mI (2403) wifi: state: 0 -> 2 (b0)[0m
    [0;32mI (2410) wifi: state: 2 -> 3 (0)[0m
    [0;32mI (2417) wifi: state: 3 -> 5 (10)[0m
    [0;32mI (2421) wifi: pm start, type: 0[0m

    [19:59:45.037]ÊÕ¡û¡ô[0;32mI (2549) ctlrConfig: SYSTEM_EVENT_STA_CONNECTED[0m

    [19:59:46.036]ÊÕ¡û¡ô[0;32mI (3547) event: sta ip: 192.168.123.90, mask: 255.255.255.0, gw: 192.168.123.1[0m
    [0;32mI (3592) ctlrConfig: SYSTEM_EVENT_STA_GOT_IP,heap size:39112[0m
    [0;32mI (3553) ctlrConfig: binded,mqtt start[0m

    mqtt_main|553 :: mqtt example
    mqtt_main|578 :: mqtt_params.host:******.com
    [1;35m[wrn] IOT_MQTT_Construct(485): Using default port: [443]
    [0m[1;35m[wrn] IOT_MQTT_Construct(492): Using default client_id: ******.******|timestamp=2524608000000,securemode=2,signmethod=hmacsha256,gw=0,ext=0,_v=sdk-c-3.0.1|
    [0m[1;35m[wrn] IOT_MQTT_Construct(499): Using default username: ******&******
    [0m[1;35m[wrn] IOT_MQTT_Construct(507): Using default password: ******
    [0m[1;35m[wrn] IOT_MQTT_Construct(520): Using default request_timeout_ms: 2000, configured value(0) out of [500, 5000]
    [0m[1;35m[wrn] IOT_MQTT_Construct(535): Using default keepalive_interval_ms: 60000, configured value(0) out of [30000, 180000]
    [0m[1;35m[wrn] IOT_MQTT_Construct(541): Using default read_buf_size: 1024
    [0m[1;35m[wrn] IOT_MQTT_Construct(547): Using default write_buf_size: 1024
    [0m[1;33m[inf] iotx_mc_init(230): MQTT init success!
    [0m[1;36m[dbg] wrapper_mqtt_connect(2697): calling TCP or TLS connect HAL for [1/3] iteration
    [0m
    [19:59:56.760]ÊÕ¡û¡ô[0;31mE (14270) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14276) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14281) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m

    [0;31mE (14288) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14298) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14308) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14318) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14328) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14337) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14347) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14357) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14367) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14377) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14387) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14397) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14407) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    [0;31mE (14416) Dynamic Impl: mbedtls_ssl_fetch_input error=-0x7280[0m
    ....

Is there are some way to finish this error to let me not cut the power.

closed 2021-02-20 by sschors

sschors, 2021-02-20:
[fix](https://github.com/espressif/ESP8266_RTOS_SDK/commit/f54641d7f3b28adf752b3f593c6ca34a60ba56b2)

lawishere, 2021-05-26:
> [fix](https://github.com/espressif/ESP8266_RTOS_SDK/commit/f54641d7f3b28adf752b3f593c6ca34a60ba56b2)

i met the same issue, how to fix it, my bro?

2898, 2024-06-19:
First, use tools such as mqtt.fx to confirm that your certificate is correct. 
Then exclude your code, as there may be code conflicts. 
Also, it is recommended to use a newer version of the mqtt proxy