espressif/ESP8266_RTOS_SDK #1049: wolfssl 链接失败

imhmao, 2021-02-05:
msys32 + xtensa-lx106-elf  8.4.0
wolfssl examples\wolfssl_client 编译通不过
```
d:/msys32/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld.exe: D:/msys32/home/esp/ESP8266_RTOS_SDK/components/esp-wolfssl/wolfssl/lib\libwolfssl.a(integer.o).literal.mp_read_radix+0x0): undefined reference to `__ctype_ptr__'
d:/msys32/opt/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld.exe: D:/msys32/home/esp/ESP8266_RTOS_SDK/components/esp-wolfssl/wolfssl/lib\libwolfssl.a(internal.o).literal.MatchDomainName+0x0): undefined reference to `__ctype_ptr__'
collect2.exe: error: ld returned 1 exit status
make: *** [D:\msys32\home\esp\ESP8266_RTOS_SDK/make/project.mk:5
```

mmorawiec, 2021-03-02:
I got the same linker errors. It seems that the wolfssl was built using old toolchain 4.8.5, but should be rebuilt with new toolchain 8.4.0.
The old toolchain contains the missing symbol:
```
$ xtensa-lx106-elf-nm -C --defined-only -g xtensa-lx106-elf-4.8.5/xtensa-lx106-elf/lib/libc.a | grep __ctype_ptr__ || echo not found
00000000 D __ctype_ptr__
```
but new one does not:
```
$ xtensa-lx106-elf-nm -C --defined-only -g xtensa-lx106-elf/xtensa-lx106-elf/lib/libc.a | grep __ctype_ptr__ || echo not found
not found
```

Could you please take a look at it and check if wolfssl binary rebuilt with new toolchain is required? I guess new binary can only be provided by espressif.
I also tried to take library from esp-wolfssl repo which seems to be newer, but the integer.o fails as well.

hellozed, 2021-03-10:
Same problem happens on Mac system.

Could anyone from Expressif help to commit a bug fix for this

KonssnoK, 2021-09-05:
I solved by implementing my own CTYPE, which wolfSSL allows you to do:

in user_settings.h

`#define CTYPE_USER
#define XTOLOWER toLower
#define XTOUPPER toUpper`

zzzzccccbbbb, 2022-06-07:
hi,friend,i try your idea,but the question is also exist.