espressif/ESP8266_RTOS_SDK #1061: I2S example does not work

k-a-z-u, 2021-03-06:
## Environment
 - Development Kit: Wemos D1 mini
 - IDF Version:	89a3f254
 - Development Env: make
 - Operating System: OpenSuse Linux

## Problem Description
The provided I2S example does not seem to work.
i2s_write(..) always blocks for the whole ticks_to_wait and i2s_bytes_write is 0 hereafter.

### Expected Behavior
working example, i2s_bytes_write > 0

### Actual Behavior
not working example, i2s_bytes_write = 0

### Steps to repropduce

	Download Toolchain
		wget https://dl.espressif.com/dl/xtensa-lx106-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz
	
	Download SDK
		Tried both:
		git clone https://github.com/espressif/ESP8266_RTOS_SDK.git
		and
		git clone --branch "release/v3.4" https://github.com/espressif/ESP8266_RTOS_SDK.git

	cd into ESP8266_RTOS_SDK/examples/peripherals/i2s/

	In main/i2s_example_main.c, add one line below line 69:
		printf("%d\n", i2s_bytes_write);
		
	make flash
	
	
## Also

The source code and documentation are unclear whether i2s_pin_config_t
just enables/disables the pins, or assigns actual pin numbers, like within the ESP-IDF

mutje, 2021-04-08:
I have a same problem. Also tried on Ai-thinker ESP-01 module.
I also measured U0RXD pin (should be I2SO_DATA) with Saleae Logic analyzer, it's always LOW (0 V).
I would be very pleased if there was at least a workaround.

kangear, 2023-03-31:
same problem(ESP-12S)