espressif/ESP8266_RTOS_SDK #1064: sdk v3.4-rc does not work!

hellozed, 2021-03-16:
The printf() sprintf() and snprintf() in newlib does not support "float", "double" and "long long" data types, this is horrible. Can we have the options of choosing lib file back?

This version SDK suggests using the v8.4.0 toolchain, but if you turn on the wolfssl, you will see many errors caused by the SDK library code linked with v.5.2.0 toolchain. It looks like this "release candidate" has not been tested on the v8.4.0 toolchain at all.

Could anybody from Exprssif help? Thanks.

dsptech, 2021-03-17:
Could this be useful?  (If I remember correctly, the related setting should be available also in the menuconfig)

https://github.com/espressif/ESP8266_RTOS_SDK/issues/995

hellozed, 2021-03-17:
Thanks, dsptech. 

The option for setting nano library was removed from the v3.4 SDK. If I turn on "nano library", then the compiler returns an error like: 

"xtensa-lx106-elf/bin/ld: cannot find -lc_nano
collect2: error: ld returned 1 exit status"

dsptech, 2021-03-19:
It seem to me a bit strange. I've the lastest update of the master branch (aka 3.4). 
There is a "newlib" option inside "menuconfig->componen config".
Inside "newlib" there is an additional "nano" option that must be unchecked to have full printf features ... and a compiled size increased of 50KB too. 
I've no linker errors at all.

hellozed, 2021-03-19:
Thanks, dsptech.

in the v3.3 sdk:
nano has options like "normal, nano, float nano, customer". The "float nano" works perfectly for me.

in the v3.4 sdk, the above options are gone. Only can either turn on nano, or turn it off. 
Actually, I have to turn it off because there is no nano library file actually. 
If I unchecked the "nano", my print()/nsprintf() cannot handle the float data.

hellozed, 2021-03-19:
by the way, are you using the v5.2.0 toolchain or v8.4.0?

I am using v5.2.0 because v8.4.0 causes compiling errors on wolfssl.

dsptech, 2021-03-19:
Toolchain path: /opt/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: esp-2020r3-49-gd5524c1
Compiler version: 8.4.0

I've no info about wolfssl.

dsptech, 2021-03-19:
could you run "git log" and "git status" to see if your sdk is up to date?

hellozed, 2021-03-19:
It looks like I have the latest sdk. 

I modified several files to relocate the library from IRAM to flash because IRAM size issue, also switch to a newer JSON library. not sure if these have an impact on the print(), I will double-check later.

thanks.

==========================
$git log

commit 89a3f254b63819035f65d9c5dcdae8864f1a6a8a (HEAD, tag: v3.4-rc, origin/release/v3.4, origin/master, origin/HEAD, release/v3.4, master)
Merge: f6c17cb2 89ab3250
Author: Dong Heng <dongheng@espressif.com>
Date:   Fri Jan 22 14:05:47 2021 +0800

    Merge branch 'bugfix/fix_rf_state_error_when_scan_again_in_scan_cb' into 'master'

    fix(lib): fix rf state error when scan again in scan cb

    See merge request sdk/ESP8266_RTOS_SDK!1588

========================

$ git status
HEAD detached at v3.4-rc
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
	modified:   components/esp8266/ld/esp8266_fragments.lf
	modified:   components/esp8266/linker.lf
	modified:   components/json/cJSON (new commits, modified content)

no changes added to commit (use "git add" and/or "git commit -a")

dsptech, 2021-03-20:
Relating your linker error, I searched deep inside the compiler directories (v8.4.0 - windows version). The library files exists (see the capture). Have you searched about the missing library ?.

[image]