espressif/ESP8266_RTOS_SDK #1065: xtensa-lx106-elf/lib/crt0.o:(.literal+0x0): undefined reference to `main'

BbIKTOP, 2021-03-16:
Have installed following [this guide](xtensa-lx106-elf/lib/crt0.o:(.literal+0x0): undefined reference to `main'). When doing make got the error
```
viktor@Work:~/work/esp8266/hello_world $ make
Toolchain path: /Users/viktor/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: esp-2020r3-49-gd5524c1
Compiler version: 8.4.0
Python requirements from /Users/viktor/esp8266/ESP8266_RTOS_SDK/requirements.txt are satisfied.
Building partitions from /Users/viktor/esp8266/ESP8266_RTOS_SDK/components/partition_table/partitions_singleapp.csv...
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/main/bootloader_start.o
AR /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/main/libmain.a
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/src/bootloader_clock.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/src/bootloader_common.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/src/bootloader_flash.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/src/bootloader_init.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/src/bootloader_random.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/src/bootloader_sha.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/src/bootloader_utility.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/src/efuse.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/src/esp_image_format.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/src/flash_encrypt.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/src/flash_partitions.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/src/flash_qio_mode.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/src/secure_boot.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/src/secure_boot_signatures.o
AR /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader_support/libbootloader_support.a
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/spi_flash/src/spi_flash.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/spi_flash/src/spi_flash_raw.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/spi_flash/port/port.o
AR /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/spi_flash/libspi_flash.a
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/log/log.o
AR /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/log/liblog.a
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/esp8266/source/ets_printf.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/esp8266/source/crc.o
CC /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/esp8266/source/esp_fast_boot.o
AR /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/esp8266/libesp8266.a
LD /Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader.elf
/Users/viktor/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /Users/viktor/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib/crt0.o:(.literal+0x0): undefined reference to `main'
collect2: error: ld returned 1 exit status
make[1]: *** [/Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader.elf] Error 1
make: *** [/Users/viktor/Documents/work/esp8266/hello_world/build/bootloader/bootloader.bin] Error 2
viktor@Work:~/work/esp8266/hello_world $
```

Macos 11.2.3

BbIKTOP, 2021-03-16:
Sorry, got an extra flags in the .profile

closed 2021-03-16 by BbIKTOP

eadronlin, 2021-12-09:
> Sorry, got an extra flags in the .profile

@BbIKTOP I've got the same error.How did you fix it?

ErhardSiegl, 2022-02-02:
I had the same error. The reason was that I had `export LDFLAGS=...`  in my profile.

clibing, 2023-05-25:
> I had the same error. The reason was that I had `export LDFLAGS=...` in my profile.

echo $LDFLAGS
-L/usr/local/opt/ruby/lib

Comment on this line

This line is a prompt for installing Python using brew. Please comment it out