espressif/ESP8266_RTOS_SDK #1074: SDK v3.4-rc can not work under -DNDEBUG

windowsair, 2021-04-05:
## Environment

- IDF version   sdk v3.4rc (89a3f254)
- Operating System:     WSL1

## Problem Description

If you try to set the Assertion Level to Disabled(-DNDEBUG) in menuconfig, you will get strange error output when generating the bin and terminate as a result.

### Expected Behavior
See above

### Actual Behavior
See above

### Steps to repropduce

1. Start with any project.  Like examples/get-started/hello_world

2. Change Assertion Level to `Disabled`
[image]

3.  Run `idf.py build`. Then see Debug Logs

## Debug Logs

```
[589/589] Generating binary image from built executable
FAILED: .bin_timestamp
cd /mnt/g/work/esp8266/sdkV3.4rc/examples/get-started/hello_world/build && /root/.espressif/python_env/rtos3.4_py2.7_env/bin/python /mnt/g/work/esp8266/sdkV3.4rc/components/esptool_py/esptool/esptool.py --chip esp8266 elf2image --flash_mode dio --flash_freq 40m --flash_size 2MB --version=3 -o /mnt/g/work/esp8266/sdkV3.4rc/examples/get-started/hello_world/build/hello-world.bin hello-world.elf && /usr/bin/cmake -E echo "Generated /mnt/g/work/esp8266/sdkV3.4rc/examples/get-started/hello_world/build/hello-world.bin" && /usr/bin/cmake -E md5sum /mnt/g/work/esp8266/sdkV3.4rc/examples/get-started/hello_world/build/hello-world.bin > /mnt/g/work/esp8266/sdkV3.4rc/examples/get-started/hello_world/build/.bin_timestamp
esptool.py v2.4.0
.flash.rodata len 0x02f3c load 0x4021d5c0

A fatal error occurred: Segment loaded at 0x4021d5c0 lands in same 64KB flash mapping as segment loaded at 0x40210010. Can't generate binary. Suggest changing linker script or ELF to merge sections.
```

windowsair, 2021-04-06:
OK, I tested it on the latest mainline and the problem no longer occurs.

windowsair, 2021-04-24:
Unfortunately, this issue is still present on the v3.4 release....

q1902505163, 2021-07-07:
up ，这个问题解决了吗？我也碰到了

windowsair, 2021-07-07:
> up ，这个问题解决了吗？我也碰到了

没有，只能暂时改成别的选项

q1902505163, 2021-07-07:
那你现在是改了什么才能跑起来

windowsair, 2021-07-07:
> 那你现在是改了什么才能跑起来

看上面的图，只要不用disable就能跑

q1902505163, 2021-07-07:
我这样的话 后面会一直crash。

windowsair, 2021-07-07:
看看core dump，是不是自己代码的问题

q1902505163, 2021-07-07:
好的，谢谢你的回复

q1902505163, 2021-07-07:
up 有联系方式吗？我有问题想要请教

windowsair, 2021-07-07:
我不是乐鑫的fae呀。。有问题你可以直接issue上提

amakukha, 2021-07-22:
I have similar issue when migrating my project from v3.3 to v3.4, even with assertion level disabled.

lingge, 2021-07-27:
我两台电脑，一台可以，一台不行，NND一样的SDK一样的代码。

mgottschlag, 2022-10-21:
I bisected the problem with the following result:

28f466d12eb00ba8569cc11188a94547a5f8e913 is the first bad commit
commit 28f466d12eb00ba8569cc11188a94547a5f8e913
Author: donghengqaz <930490596@qq.com>
Date:   Wed Feb 12 10:22:49 2020 +0800

    feat(gcc): Update cross toolchain GCC to v8.x

mgottschlag, 2022-10-21:
The following line introduced by the referenced commit is broken because esp_newlib_init() is not executed in NDEBUG builds:

    assert(esp_newlib_init() == 0);