espressif/ESP8266_RTOS_SDK #1082: Disable log output

vyacheslav-shubin, 2021-04-17:
How to disable log output?

int log_handler(int ch) {
    return 0;
}

void app_main() {
    esp_log_set_putchar(log_handler);
    esp_log_level_set("wifi", ESP_LOG_NONE);

Has no effect.

I (4238) wifi:state: 0 -> 2 (b0)
I (4285) wifi:state: 2 -> 3 (0)
I (4291) wifi:state: 3 -> 5 (10)
I (14290) wifi:state: 5 -> 0 (cc00)
I (16470) wifi:state: 0 -> 2 (b0)
I (16508) wifi:state: 2 -> 3 (0)
I (16514) wifi:state: 3 -> 5 (10)
I (16547) wifi:connected with ******, aid = 1, channel 6, HT20, bssid = ******
I (81655) wifi:bcn_timout,ap_probe_send_start

zheyuejiang, 2021-04-22:
please show your menuconfig log outpt settings here

vyacheslav-shubin, 2021-04-22:
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_TARGET_ESP8266=y
CONFIG_IDF_TARGET="esp8266"

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-lx106-elf-"
CONFIG_SDK_PYTHON="python"
# CONFIG_SDK_MAKE_WARN_UNDEFINED_VARIABLES is not set
# CONFIG_BOOTLOADER_INIT_SPI_FLASH is not set
# CONFIG_BOOTLOADER_DISABLE_JTAG_IO is not set
# CONFIG_BOOTLOADER_FAST_BOOT is not set
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=y
# CONFIG_LOG_BOOTLOADER_LEVEL_ERROR is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_WARN is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_INFO is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE is not set
CONFIG_LOG_BOOTLOADER_LEVEL=0
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_STORE_OFFSET=0x0
CONFIG_ESPTOOLPY_PORT="/home/shubin/tty/ttyWIFI"
CONFIG_ESPTOOLPY_BAUD_115200B=y
# CONFIG_ESPTOOLPY_BAUD_230400B is not set
# CONFIG_ESPTOOLPY_BAUD_921600B is not set
# CONFIG_ESPTOOLPY_BAUD_2MB is not set
# CONFIG_ESPTOOLPY_BAUD_OTHER is not set
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=115200
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
# CONFIG_ESPTOOLPY_FLASHMODE_QOUT is not set
# CONFIG_ESPTOOLPY_FLASHMODE_DIO is not set
# CONFIG_ESPTOOLPY_FLASHMODE_DOUT is not set
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_SPI_FLASH_MODE=0x0
# CONFIG_ESPTOOLPY_FLASHFREQ_80M is not set
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
# CONFIG_ESPTOOLPY_FLASHFREQ_26M is not set
# CONFIG_ESPTOOLPY_FLASHFREQ_20M is not set
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
CONFIG_SPI_FLASH_FREQ=0x0
# CONFIG_ESPTOOLPY_FLASHSIZE_1MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=y
# CONFIG_ESPTOOLPY_FLASHSIZE_4MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_8MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_16MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE="2MB"
CONFIG_SPI_FLASH_SIZE=0x200000
CONFIG_ESPTOOLPY_BEFORE_RESET=y
# CONFIG_ESPTOOLPY_BEFORE_NORESET is not set
CONFIG_ESPTOOLPY_BEFORE="default_reset"
CONFIG_ESPTOOLPY_AFTER_HARD_RESET=y
# CONFIG_ESPTOOLPY_AFTER_SOFT_RESET is not set
# CONFIG_ESPTOOLPY_AFTER_NORESET is not set
CONFIG_ESPTOOLPY_AFTER="hard_reset"
# CONFIG_ESPTOOLPY_MONITOR_BAUD_9600B is not set
# CONFIG_ESPTOOLPY_MONITOR_BAUD_57600B is not set
CONFIG_ESPTOOLPY_MONITOR_BAUD_74880B=y
# CONFIG_ESPTOOLPY_MONITOR_BAUD_115200B is not set
# CONFIG_ESPTOOLPY_MONITOR_BAUD_230400B is not set
# CONFIG_ESPTOOLPY_MONITOR_BAUD_921600B is not set
# CONFIG_ESPTOOLPY_MONITOR_BAUD_2MB is not set
# CONFIG_ESPTOOLPY_MONITOR_BAUD_OTHER is not set
CONFIG_ESPTOOLPY_MONITOR_BAUD_OTHER_VAL=74880
CONFIG_ESPTOOLPY_MONITOR_BAUD=74880
[... 429 lines cut ...]
CONFIG_TCP_MAXRTX=12
CONFIG_TCP_SYNMAXRTX=6
CONFIG_TCP_MSS=1440
CONFIG_TCP_MSL=60000
CONFIG_TCP_SND_BUF_DEFAULT=2880
CONFIG_TCP_WND_DEFAULT=5760
CONFIG_TCP_RECVMBOX_SIZE=6
CONFIG_TCP_QUEUE_OOSEQ=y
# CONFIG_ESP_TCP_KEEP_CONNECTION_WHEN_IP_CHANGES is not set
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=2048
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
# End of deprecated options

swanav, 2021-04-30:
I notice that you have CONFIG_LOG_DEFAULT_LEVEL=3 in the menuconfig. 

Try setting the 
```
CONFIG_LOG_DEFAULT_LEVEL_NONE=y
CONFIG_LOG_DEFAULT_LEVEL=0
```

This should do it!

vyacheslav-shubin, 2021-05-13:
Nothing changed. 

CONFIG_LOG_DEFAULT_LEVEL_NONE=y
CONFIG_LOG_DEFAULT_LEVEL=0

I (5504) wifi:state: 0 -> 2 (b0)
I (5545) wifi:state: 2 -> 3 (0)
I (5551) wifi:state: 3 -> 5 (10)
I (15549) wifi:state: 5 -> 0 (cc00)
I (17730) wifi:state: 0 -> 2 (b0)
I (17769) wifi:state: 2 -> 3 (0)
I (17773) wifi:state: 3 -> 5 (10)

iot_sdk have a function for disable system logging system_set_os_print(bool enable_outpur)
Perhaps the same is in this SDK?

Sonic-Amiga, 2023-06-24:
Very nice support from Espressif 👎

Sonic-Amiga, 2023-06-24:
The code which does this, is located inside libnet80211.a . Well, i love reverse engineering.

Sonic-Amiga, 2023-06-24:
I've come up with a solution. The wi-fi code for some strange reason uses boot-time logging functions, which bypass all the common controls. Fortunately Espressif guys have hit this problem too, so they implemented an internal flag, which allows to suppress ets_printf() function. See [here](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/esp8266/source/ets_printf.c#L294) for details
```
extern uint8_t FlashIsOnGoing;

[skip]

   // Shut off console log if it is using our UART
   if (uart == CONFIG_ESP_CONSOLE_UART_NUM) {
      FlashIsOnGoing = 1;
      // Wait for previously emitted log text to reach its destination
      // fflush(stdout) doesn't do the job
      uart_wait_tx_done(uart, 1000 / portTICK_PERIOD_MS);
      // Only after this we switch the output, otherwise we lose the final line.
      // Apparently stdout does tons of buffering together with the UART driver
      esp_log_set_putchar(faikin_log_putc);
   }
```
 I guess, the main question is: why is wi-fi daemon even using boot-time logger ? Looks like a bug.

Sonic-Amiga, 2023-06-24:
And yes, the only legitimate way to disable this is to set CONFIG_ESP_CONSOLE_UART_NONE , which would kill all the logs completely by turning uart_putc() into a NOP

Sonic-Amiga, 2023-09-24:
After commit 69cf000a4aab6128fb268d99c2665700122f6c67 my solution seems to stop working. Perhaps they switched to ets_vprintf(), which isn't suppressed at all

@zhangyanjiaoesp Hello! Are you reading this ? I thought the project is dead in the water

Sonic-Amiga, 2023-09-25:
Unfortunately the solution employing an existing FlashIsOnGoing flag did not go well, because every NVS write clears it back to zero. I had to make my own: https://github.com/espressif/ESP8266_RTOS_SDK/pull/1253