espressif/ESP8266_RTOS_SDK #1083: [v3.4] Undefined reference to `pthread_...` functions

yhsb2k, 2021-04-18:
- Development Kit: esp8266 nodemcu board
- IDF version: v3.4 (89a3f254)
- Development Env: cmake
- Operating System: Windows
- Power Supply: USB

## Problem Description
I am building ESP8266_RTOS_SDK as static library embedded to the project.
I haven't made any changes in ESP8266_RTOS_SDK.

Unfortunately, I get build error during linking:
```log
c:/mydevtools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld.exe: c:/mydevtools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib\libstdc++.a(guard.o):(.literal._ZN12_GLOBAL__N_1L16init_static_condEv+0x8): undefined reference to `pthread_cond_init'
c:/mydevtools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld.exe: c:/mydevtools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib\libstdc++.a(guard.o):(.literal._ZN9__gnu_cxx6__cond9broadcastEv[__gnu_cxx::__cond::broadcast()]+0x8): undefined reference to `pthread_cond_broadcast'
c:/mydevtools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld.exe: c:/mydevtools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib\libstdc++.a(guard.o):(.literal.__cxa_guard_acquire+0x8): undefined reference to `pthread_cond_wait'
c:/mydevtools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld.exe: c:/mydevtools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib\libstdc++.a(guard.o): in function `__gthread_cond_init_function':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/xtensa-lx106-elf/bits/gthr-default.h:858: undefined reference to `pthread_cond_init'
c:/mydevtools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld.exe: c:/mydevtools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib\libstdc++.a(guard.o): in function `__gnu_cxx::__cond::broadcast()':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/ext/concurrence.h:280: undefined reference to `pthread_cond_broadcast'
c:/mydevtools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld.exe: c:/mydevtools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib\libstdc++.a(guard.o): in function `__gthread_cond_wait':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/xtensa-lx106-elf/bits/gthr-default.h:877: undefined reference to `pthread_cond_wait'
collect2.exe: error: ld returned 1 exit status
mingw32-make.exe[3]: *** [CMakeFiles\omef-esp8266.dir\build.make:185: omef-esp8266] Error 1
mingw32-make.exe[2]: *** [CMakeFiles\Makefile2:2166: CMakeFiles/omef-esp8266.dir/all] Error 2
mingw32-make.exe[1]: *** [Makefile:148: all] Error 2
make: *** [makefile:13: all] Error 2
```
See the full build log: [build.log](https://github.com/espressif/ESP8266_RTOS_SDK/files/6331682/build.log)

The following code in main.cpp fixes the issue, but this is **not the proper fix**:
```c++
extern "C" void pthread_cond_init(void) {}
extern "C" void pthread_cond_broadcast(void) {}
extern "C" void pthread_cond_wait(void) {}
```
*Interesting things that source file with these functions (components/pthread/pthread_cond_var.c) is compiled and linked, but somehow the symbols above are not found by linker.*

### Expected Behavior
Build OK.

### Actual Behavior
Build fails.

### Steps to repropduce
Use the following simple cmake file:
```cmake
cmake_minimum_required(VERSION 3.13)

set(CMAKE_TOOLCHAIN_FILE src/hal/ESP8266/ESP8266_RTOS_SDK/tools/cmake/toolchain-esp8266.cmake)

project(omef-esp8266 C CXX ASM)

set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

set(CMAKE_CXX_STANDARD 17)
if(NOT CMAKE_BUILD_TYPE AND NOT CMAKE_CONFIGURATION_TYPES)
    set(CMAKE_BUILD_TYPE Debug)
endif()

add_compile_options(
    $<$<NOT:$<COMPILE_LANGUAGE:ASM>>:-mlongcalls>
)

include_directories(./src)

# ESP8266_RTOS_SDK
include(src/hal/ESP8266/ESP8266_RTOS_SDK/tools/cmake/idf.cmake)
idf_build_process(esp8266
    SDKCONFIG ${CMAKE_BINARY_DIR}/sdkconfig
    COMPONENTS esp8266 freertos esptool_py pthread spi_flash
)

add_executable(${CMAKE_PROJECT_NAME} src/main.cpp)

target_link_options(${CMAKE_PROJECT_NAME} PRIVATE
    ${CPU_FLAGS}
    -nostdlib
    -Wl,--gc-sections
    -Wl,-Map=${CMAKE_PROJECT_NAME}.map,--cref
)

target_link_libraries(${CMAKE_PROJECT_NAME} PRIVATE
    idf::esp8266
    idf::freertos
    idf::pthread
    idf::spi_flash
)
idf_build_executable(${CMAKE_PROJECT_NAME})
```

amakukha, 2021-07-23:
I have same issue. Even the examples sometimes produce this problem. But, unfortunately, the workaround doesn't work...

herve-sg, 2021-08-17:
There is a mistake in the CMakeLists.txt file of component "pthread".

You and patch that for correct the linker of linking file "pthread_cond_var.c"
```
  set(extra_link_flags "-u pthread_include_pthread_impl")
- list(APPEND extra_link_flags "-u pthread_include_pthread_cond_impl")
+list(APPEND extra_link_flags "-u pthread_include_pthread_cond_var_impl")
  list(APPEND extra_link_flags "-u pthread_include_pthread_local_storage_impl")
```

yhsb2k, 2021-11-27:
Thank you @herve-sg!

Additionally to CMakeLists.txt, need to change components/pthread/component.mk:
```diff
diff --git a/components/pthread/component.mk b/components/pthread/component.mk
index c5793fde..0e1cd8a3 100644
--- a/components/pthread/component.mk
+++ b/components/pthread/component.mk
@@ -16,6 +16,6 @@ ifeq ($(GCC_NOT_5_2_0), 1)
 # Forces the linker to include pthread implementation from this component,
 # instead of the weak implementations provided by libgcc.
 COMPONENT_ADD_LDFLAGS += -u pthread_include_pthread_impl
-COMPONENT_ADD_LDFLAGS += -u pthread_include_pthread_cond_impl
+COMPONENT_ADD_LDFLAGS += -u pthread_include_pthread_cond_var_impl
 COMPONENT_ADD_LDFLAGS += -u pthread_include_pthread_local_storage_impl
 endif  # GCC_NOT_5_2_0
 
```