espressif/ESP8266_RTOS_SDK #1084: I have issue  with make flash

bocheta, 2021-04-22:
Hello everyone,
when i execute make flash shows me errors, i dont know how to solve it. Any idea. thank you

[image]

zheyuejiang, 2021-04-22:
toolchain path is not correct

zheyuejiang, 2021-04-22:
your current toolchain is ESP32 not ESP8266, make sure what do you want to compile

bocheta, 2021-04-22:
> your current toolchain is ESP32 not ESP8266, make sure what do you want to compile

Hi zheyuejiang
i did the same thing as mentioned in this link https://docs.espressif.com/projects/esp8266-rtos-sdk/en/v3.4/get-started/linux-setup.html#
what should i do exactelly to remove this error. Thank you

zheyuejiang, 2021-04-22:
just follow the Toolchain Setup is OK.
Plz show your PATH here .

bocheta, 2021-04-22:
> just follow the Toolchain Setup is OK.
> Plz show your PATH here .

[image]

zheyuejiang, 2021-04-22:
your IDF_PATH is error, esp-idf is esp32

zheyuejiang, 2021-04-22:
try 
/home/anasse/esp/ESP8266_RTOS_SDK

bocheta, 2021-04-22:
> try
> /home/anasse/esp/ESP8266_RTOS_SDK

i did it actually but does not working.
i did also like this instruction
[image]

zheyuejiang, 2021-04-22:
Do you have try the script under esp/ESP8266_RTOS_SDK/export.sh ?
The script will auto process the IDF_PATH, and then the IDF_PATH will be like /home/anasse/esp/ESP8266_RTOS_SDK

bocheta, 2021-04-22:
> Do you have try the script under esp/ESP8266_RTOS_SDK/export.sh ?
> The script will auto process the IDF_PATH, and then the IDF_PATH will be like /home/anasse/esp/ESP8266_RTOS_SDK

i did it works thank you. Then i run make flash and it crashed.
[image]

zheyuejiang, 2021-04-22:
It just can't open the /dev/ttyUSB0 port.
plz check the  /dev/ttyUSB0 port availiable

bocheta, 2021-04-22:
> It just can't open the /dev/ttyUSB0 port.
> plz check the /dev/ttyUSB0 port availiable

Thank you, i enabled the usb port. Then shows me this after i run make flash
[image]

zheyuejiang, 2021-04-22:
plz check chip is waiting for downloading

bocheta, 2021-04-22:
> plz check chip is waiting for downloading
What should i do in chip please?
Do i have to press any button?