espressif/ESP8266_RTOS_SDK #1089: There was an error with the wifi scan

GarMingLi, 2021-04-27:
## Environment
- IDF version : 18c3a1944ac3b1454a87e578705cd90026525608
- Development Env:      Make
- Operating System:     win10 wsl（Ubuntu18.04）
- toolchain: xtensa-lx106-elf-linux32-1.22.0-100-ge567ec7-5.2.0.tar
- Board: NodeMCU

## Problem Description
Wifi mode runs in AP/STA mode.Wifi scans are performed every 3 minutes.However, the following errors often occur that cause a restart：
```
ESP_ERROR_CHECK failed: esp_err_t 0x3006 (ESP_ERR_WIFI_STATE) at 0x40221574
file: "user_wifi.c" line 458
func: wifi_scan_timer_cb
expression: esp_wifi_scan_start(&configForScan, false)

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 7328, room 16 
tail 0
chksum 0xe4
load 0x3ffe8408, len 24, room 8 
tail 0
chksum 0x0d
load 0x3ffe8420, len 3544, room 8 
tail 0
```
Code for wifi scanning:
```
static void wifi_scan_timer_cb(TimerHandle_t xTimer) {
    ESP_LOGI(TAG, "wifi_scan_timer_cb");

    if (!wifi_scan_flag) {
        wifi_scan_flag = true;
        esp_wifi_scan_stop();
    }
    ESP_ERROR_CHECK(esp_wifi_scan_start(&configForScan, false));
}

```

GarMingLi, 2021-04-30:
How to solve this problem? I tried 'ESP_ERROR_CHECK()'function cancels and judges the return value. If it fails, disconnect the WiFi connection and scan again. But the experience is not very good.

dsptech, 2021-05-05:
Hi, suggest to check also the code returned by esp_wifi_scan_stop().
On another hand, if I remember correctly, you have to make sure the releasing of any resource allocated by the previous esp_wifi_scan_start(...) call. 
To release those resources is required a final call to esp_wifi_scan_get_ap_records(...) (esp_wifi_scan_stop does not perform that).

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html