espressif/ESP8266_RTOS_SDK #1090: Ho do I enable insecure TLS connection

janav1, 2021-04-27:
## Environment

- Development Kit: ESP-WROOM02
- IDF version: v3.4 89a3f254b63819035f65d9c5dcdae8864f1a6a8a
- Development Env:      Eclipse IDE + Make
- Operating System:     Windows 10 x64
- Power Supply:         external 5V

## Problem Description
I'm trying to enable TLS encryption for MQTT and HTTP protocol without certificate verification. However, both the MQTT and HTTP_Client libraries fail (log below). There should be an option "CONFIG_ESP_TLS_INSECURE" to bypass certificate checking, but it is missing in MENUCONFIG for ESP8266. How to achieve an encrypted connection to the server without providing a certificate? 

### Expected Behavior
Connect to server even without checking the certificate (insecure way).

### Actual Behavior
Connection fails.

### Steps to repropduce
Just set the module to WiFi Client (STA) mode, connect to your router and use code below.

### Code to reproduce this issue
_http_event_handler is the default one provided in examples
```cpp
char *post_data = malloc(512);
esp_http_client_config_t config = {
            .host = "https://www.howsmyssl.com",
            .path = "/a/check",
            .transport_type = HTTP_TRANSPORT_OVER_SSL,
            .event_handler = _http_event_handler,
            .cert_pem = NULL // <---
      };

      esp_http_client_handle_t client = esp_http_client_init(&config);
      esp_err_t err = esp_http_client_perform(client);
      if (err == ESP_OK) {
         int sc = esp_http_client_get_status_code(client);
         ESP_LOGI(TAG, "HTTPS Status = %d, content_length = %d", sc, esp_http_client_get_content_length(client));
         if(sc == 200){
            esp_http_client_read_response(client, post_data, 512);
         }
      }
      else {
         ESP_LOGE(TAG, "Error perform http request %s", esp_err_to_name(err));
      }
      esp_http_client_cleanup(client);
      free(post_data);
```

## Debug Logs
```
<ESC>[0;31mE (127251) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x4c<ESC>[0m<CR><LF>
<ESC>[0;31mE (127255) esp-tls: Failed to open new connection<ESC>[0m<CR><LF>
<ESC>[0;31mE (127258) TRANS_SSL: Failed to open a new connection<ESC>[0m<CR><LF>
<ESC>[0;31mE (127269) HTTP_CLIENT: Connection failed, sock < 0<ESC>[0m<CR><LF>
<ESC>[0;31mE (127274) HTTPS: Error perform http request ESP_ERR_HTTP_CONNECT<ESC>[0m<CR><LF>
```

## Other items if possible
[sdkconfig.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/6382135/sdkconfig.txt)
[image]

ESP-YJM, 2021-04-30:
@janav1 Please use the master branch and try. We implement the option `ESP_TLS_INSECURE`.

janav1, 2021-05-03:
Thank you for implementing the option, but the issue still persist. The encrypted connection cant be established. Error is same as before. Do I have to make any other changes except to copy the master branch and enable option in menuconfig?

ESP-YJM, 2021-05-06:
@janav1 Yes, you need enable option in menuconfig(make menuconfig->component config->ESP-TLS->Enable Allow potentially insecure options)

janav1, 2021-05-06:
Both new options are enabled:
```
[*] Allow potentially insecure options
[*] Skip server certificate verification by default (WARNING: ONLY FOR TESTING PURPOSES, READ HELP)
```
but the issue still persist.

ESP-YJM, 2021-05-06:
@janav1 I think the issue you pasted in log is not the same issue you mentioned. The error `esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x4c` means `MBEDTLS_ERR_NET_RECV_FAILED`.

janav1, 2021-05-06:
The posted log is printed by the ESP module each time I try to connect securely to any server. Making same request without TLS encryption pass succesfully.
There is something strange with http_client library and used RAM. Before making secure request, free RAM is about 38 kB. After the request:
`esp_get_minimum_free_heap_size(): 2300`
so the connection takes 36 kB of RAM!
There is no other task running.

ESP-YJM, 2021-05-06:
@janav1 Yes, TLS connection will cost much RAM. You can enable dynamic mbedtls config by `make menuconfig->component config->mbedtls->Using dynamic TX/RX buffer`, also enable two new option. It will save some RAM.

janav1, 2021-05-07:
Dear YJM, thank you for your efford, but even with dynamic buffers, the connection fails with error:
`<ESC>[0;31mE (5326) HTTPS: Failed to allocate memory for output buffer<ESC>[0m<CR><LF>
Free memory: 37372 bytes, minimum: 740<CR><CR><LF>`
I am not an IT expert, but why the library needs so much RAM even when you do not need to check the certificate? With insecure option enabled, any connection should be considered valid regardless certificate sent by the server. With Arduino-ESP it works this way.
What other options do I have to successfully establish encrypted connection?

ESP-YJM, 2021-05-07:
@janav1 Could you please share your sdkconfig file?

janav1, 2021-05-07:
Latest file with new options enabled is attached.
[sdkconfig.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/6440473/sdkconfig.txt)

ESP-YJM, 2021-05-07:
Could you please modify `CONFIG_MBEDTLS_SSL_OUT_CONTENT_LEN` to 1024 and `CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN` to 8192 in make menuconfig->component config->mbedtls and try.

janav1, 2021-05-07:
Thank you, the MQTT client with TLS works now :-)
but HTTP client still have issues:
`HTTPS: Failed to allocate memory for output buffer.`
What is the difference between making encrypted connection for MQTT and for HTTP? Both use mbedtls same way I think.

ESP-YJM, 2021-05-08:
I think it is same with MQTT and HTTP. What MQTT client and HTTPS client example you used?

janav1, 2021-05-10:
HTTP and MQTT client is the one provided by Espressif with RTOS SDK v3.3.