espressif/ESP8266_RTOS_SDK #1092: esp8266 release/v2.x.x 不能编译第三方静态库【lwip】

Hessonlam, 2021-04-28:
clone 了esp8266 release/v2.x.x 或者v2.0.0 版本都不能完成 编译
提示部分函数的声明缺失.
通过grep 命令 全局搜索 该函数 发现没有此函数的说明
应该怎么解决？

下面是log
PS： build 文件夹是 sdk的output 文件夹
hesson@DESKTOP-UR7VT96:~/esp8266/ESP8266_RTOS_SDK/third_party$ sh make_lib.sh lwip
make_lib.sh version 20150924

SDK_PATH:
/home/hesson/esp8266/ESP8266_RTOS_SDK

make -C arch clean;  make -C core clean;  make -C api clean;  make -C apps clean;  make -C netif clean;
make[1]: Entering directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/arch'
rm -f -r .output/eagle/debug
make[1]: Leaving directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/arch'
make[1]: Entering directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/core'
make -C ipv4 clean;  make -C ipv6 clean;
make[2]: Entering directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/core/ipv4'
rm -f -r .output/eagle/debug
make[2]: Leaving directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/core/ipv4'
make[2]: Entering directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/core/ipv6'
rm -f -r .output/eagle/debug
make[2]: Leaving directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/core/ipv6'
rm -f -r .output/eagle/debug
make[1]: Leaving directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/core'
make[1]: Entering directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/api'
rm -f -r .output/eagle/debug
make[1]: Leaving directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/api'
make[1]: Entering directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/apps'
rm -f -r .output/eagle/debug
make[1]: Leaving directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/apps'
make[1]: Entering directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/netif'
rm -f -r .output/eagle/debug
make[1]: Leaving directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/netif'
rm -f -r .output/eagle/debug
make[1]: Entering directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/arch'
DEPEND: xtensa-lx106-elf-gcc -M -fno-aggressive-loop-optimizations -Os -g -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -fno-jump-tables -D_POSIX_SOURCE -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DMEMLEAK_DEBUG -I include -I ./ -I ../include -I ./ -I ../../include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/extra_include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/driver_lib/include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/espressif -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip/ipv4 -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip/ipv6 -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/nopoll -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/spiffs -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/ssl -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/json -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/openssl -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/mqtt sys_arch.c
xtensa-lx106-elf-gcc -fno-aggressive-loop-optimizations -Os -g -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -fno-jump-tables -D_POSIX_SOURCE -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP  -DMEMLEAK_DEBUG   -I include -I ./ -I ../include -I ./ -I ../../include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/extra_include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/driver_lib/include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/espressif -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip/ipv4 -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip/ipv6 -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/nopoll -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/spiffs -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/ssl -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/json -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/openssl -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/mqtt  -o .output/eagle/debug/obj/sys_arch.o -c sys_arch.c
xtensa-lx106-elf-ar ru .output/eagle/debug/lib/liblwiparch.a .output/eagle/debug/obj/sys_arch.o
xtensa-lx106-elf-ar: creating .output/eagle/debug/lib/liblwiparch.a
make[1]: Leaving directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/arch'
make[1]: Entering directory '/home/hesson/esp8266/ESP8266_RTOS_SDK/third_party/lwip/core'
DEPEND: xtensa-lx106-elf-gcc -M -fno-aggressive-loop-optimizations -Os -g -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -fno-jump-tables -D_POSIX_SOURCE -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DMEMLEAK_DEBUG -I include -I ./ -I ../include -I ./ -I ../../include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/extra_include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/driver_lib/include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/espressif -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip/ipv4 -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip/ipv6 -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/nopoll -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/spiffs -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/ssl -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/json -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/openssl -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/mqtt tcp_out.c
DEPEND: xtensa-lx106-elf-gcc -M -fno-aggressive-loop-optimizations -Os -g -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -fno-jump-tables -D_POSIX_SOURCE -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DMEMLEAK_DEBUG -I include -I ./ -I ../include -I ./ -I ../../include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/extra_include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/driver_lib/include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/espressif -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip/ipv4 -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip/ipv6 -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/nopoll -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/spiffs -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/ssl -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/json -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/openssl -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/mqtt udp.c
DEPEND: xtensa-lx106-elf-gcc -M -fno-aggressive-loop-optimizations -Os -g -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -fno-jump-tables -D_POSIX_SOURCE -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DMEMLEAK_DEBUG -I include -I ./ -I ../include -I ./ -I ../../include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/extra_include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/driver_lib/include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/espressif -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip/ipv4 -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/lwip/ipv6 -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/nopoll -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/spiffs -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/ssl -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/json -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/openssl -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include/mqtt def.c
DEPEND: xtensa-lx106-elf-gcc -M -fno-aggressive-loop-optimizations -Os -g -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -ffunction-sections -fdata-sections -fno-builtin-printf -fno-jump-tables -D_POSIX_SOURCE -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DMEMLEAK_DEBUG -I include -I ./ -I ../include -I ./ -I ../../include -I /home/hesson/esp8266/ESP8266_RTOS_SDK/include -I /home/hesson/esp8266/ESP826 [... cut ...]
../third_party/ssl/ssl/ssl_platform.c:                  *p ++ = system_get_data_of_array_8(base64_enc_map, ((C2 & 15) << 2) & 0x3F);
../third_party/ssl/ssl/ssl_tls1_clnt.c:        buf[offset++] = system_get_data_of_array_8(ssl_prot_prefs, i);
../third_party/ssl/ssl/ssl_tls1_svr.c:            if (system_get_data_of_array_8(ssl_prot_prefs, j) == ((buf[offset+i]<<8) + buf[offset+i+1]))   /* got a match? */
../third_party/ssl/ssl/ssl_tls1_svr.c:                ssl->cipher = system_get_data_of_array_8(ssl_prot_prefs, j);

ESP-YJM, 2021-04-30:
你需要使用 toolchain 4.8.5 来编译 lwip 库

Hessonlam, 2021-04-30:
怎么看toolchain 的版本

Hessonlam, 2021-04-30:
Toolchain 4.8.5 的版本在哪里下载呢.

ESP-YJM, 2021-04-30:
xtensa-lx106-elf-gcc --version 来查看 toolchain版本.

Hessonlam, 2021-04-30:
查看到了, 是5.2.0 的版本. 那应该在哪里下载旧版本的toolchain

ESP-YJM, 2021-04-30:
https://dl.espressif.com/dl/xtensa-lx106-elf-linux64-1.22.0-88-gde0bdc1-4.8.5.tar.gz

Hessonlam, 2021-04-30:
好的!谢谢

https://dl.espressif.com/dl/xtensa-lx106-elf-linux64-1.22.0-88-gde0bdc1-4.8.5.tar.gz