espressif/ESP8266_RTOS_SDK #1093: (ASCII, RTU, TCP) mode in Kconfig

KamranArain, 2021-05-06:
----------------------------- Delete below -----------------------------

I am trying to use serial Modbus but can't find any thing in menuconfig. if I remove tcp 
from menuconfig it generates and error 

```sh
ESP8266_RTOS_SDK/components/freemodbus/modbus/include/mbconfig.h:66:2: error: #error "None of Modbus communication mode is enabled. Please enable one of (ASCII, RTU, TCP) mode in Kconfig.
```

I also searched for `CONFIG_FMB_COMM_MODE_RTU_EN` but could not find it any where in whole project.

ESP-YJM, 2021-05-07:
@KamranArain Now ESP8266 only support Modbus TCP mode.

KamranArain, 2021-05-07:
can't I use these files [freemodbus/modbus](https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/components/freemodbus/modbus). There is RTU and ASCII file in freemodbus. There is some one who used older version v1.4.x for RTU [oh5fsz/freemodbus](https://github.com/oh5fsz/freemodbus-esp8266) I want newer version.

KamranArain, 2021-05-07:
I also set `CONFIG_FMB_COMM_MODE_RTU_EN ` in sdkconfig.h and it allowed me to play with it but I didn't succeed.

ESP-YJM, 2021-05-08:
@KamranArain Yes, [freemodbus/modbus](https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/components/freemodbus/modbus) has RTU and ASCII, but we only adapte port level to support TCP mode.