espressif/ESP8266_RTOS_SDK #1103: [v3.4] WiFi station connects and disconnects to AP repeatedly when the beacon contains multiple "Country Info".

k-tkr, 2021-06-13:
## Environment

- Development Kit: WROOM-02
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): a60ee4e1
- Development Env:      [other] Visual Studio + VisualGDB
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description
One of my access point sends beacons containing multiple "Country Info" values in the "Country Information" tag as shown below.

[image]

When connected to the AP, ESP8266 WiFi station fails to receive beacons and connects to and disconnects from the AP repeatedly unless I keep the channel fixed to the one belonging to the last "Country Info" (Ch. 13 in the above picture).
In the "badap" log below, you can see "bcn_timout,ap_probe_send_start", "send probe req", "ap_probe_send_over, rest wifi status to disassoc" messages again and again.

When I use an AP which sends beacons containing only one "Country Info" value in it, ESP keeps connected and doesn't disconnects/connects repeatedly, as "goodap" log below shows.

The described behavior doesn't happen with RTOS SDK 3.3.

### Expected Behavior
The station keeps connected to the AP.
### Actual Behavior
The station disconnects from and connects to the AP repeatedly.

### Steps to repropduce

1. Download the example code from [github](https://github.com/espressif/ESP8266_RTOS_SDK/tree/release/v3.4/examples/wifi/getting_started/station)
2.  In wifi_init_sta() in station_example_main.c, add the infinite loop after the connection (I attach the code below with sdk config);

```cpp
    if (bits & WIFI_CONNECTED_BIT) {
        ESP_LOGI(TAG, "wifi_init_sta finished.");
        ESP_LOGI(TAG,
            "connect to ap SSID:%s password:%s",
            EXAMPLE_ESP_WIFI_SSID,
            EXAMPLE_ESP_WIFI_PASS);
        wifi_ap_record_t ap_info;
        esp_wifi_sta_get_ap_info(&ap_info);
        ESP_LOGI(TAG, "ssid:%s", ap_info.ssid);
        ESP_LOGI(TAG, "primary:%d", ap_info.primary);
        ESP_LOGI(TAG, "authmode:%d", ap_info.authmode);
        int ct = 0;
        while (1) {
            ESP_LOGI(TAG, "%d", ct++);
            vTaskDelay(1000 / portTICK_RATE_MS);
        }
    }
```
3. Set ssid and password in the menu config and build the code.

### Code to reproduce this issue
[code.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/6643158/code.zip)

## Debug Logs
goodap
```
I (7402) wifi:connected with goodap, aid = 1, channel 6, HT20, bssid = xx:xx:xx:xx:xx:xx
D (7413) wifi: connect status 1 -> 5
I (8818) tcpip_adapter: sta ip: 192.168.10.25, mask: 255.255.255.0, gw: 192.168.10.1
I (8821) wifi station: got ip:192.168.10.25
I (8826) wifi station: wifi_init_sta finished.
I (8836) wifi station: connect to ap SSID:goodap password:password1
I (8850) wifi station: ssid:goodap
I (8858) wifi station: primary:6
I (8866) wifi station: authmode:3
I (8874) wifi station: 0
I (9877) wifi station: 1
I (10877) wifi station: 2
I (11877) wifi station: 3
[... 154 lines cut ...]
D (33992) wifi: rd: chan=6 active max=11
D (33999) wifi: send probe req on channel 6 bssid ff:ff:ff:ff:ff:ff ssid badap
I (34016) wifi station: retry to connect to the AP
I (34025) wifi station: connect to the AP fail
D (34134) wifi: cnx_start_handoff_cb
D (34136) wifi: cnx_connect_op
I (34138) wifi:state: 0 -> 2 (b0)
I (34142) wifi:state: 2 -> 3 (0)
I (34160) wifi:state: 3 -> 5 (10)
I (34160) wifi: aid 4
I (34162) wifi: cnt 
I (34574) wifi station: 25
D (35233) wifi: wpa_psk start
I (35237) wifi: wpa_psk handle succeed
D (35244) wifi: wpa_psk start
I (35252) wifi: wpa_psk handle succeed
I (35254) wifi:connected with badap, aid = 4, channel 6, HT20, bssid = xx:xx:xx:xx:xx:xx
D (35294) wifi: connect status 1 -> 5
I (35579) wifi station: 26
I (36450) tcpip_adapter: sta ip: 192.168.179.2, mask: 255.255.255.0, gw: 192.168.179.1
I (36453) wifi station: got ip:192.168.179.2
I (36579) wifi station: 27
I (37579) wifi station: 28
I (38579) wifi station: 29
```

mincrmatt12, 2021-09-04:
This is also impacting me: the esp8266 associates just fine and has perfectly functional network connectivity for about 3 seconds until the beacon timeout expires, at which point it reconnects. I can only assume something is rejecting beacon frames that have country information like shown here which makes the rest of the system think the AP has stopped sending them altogether -- please fix!

mincrmatt12, 2021-09-05:
This can be worked around by manually setting the country information with `esp_wifi_set_country`.