espressif/ESP8266_RTOS_SDK #1129: Outdated get-started document + running on Python3

adam-ah, 2021-08-27:
## Environment

MacOS, latest version of RTOS SDK 
Documentation from https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/index.html

## Problem Description - get-started documentation

The get-started guide is getting very outdated, does not mention installing the toolchain via `install.sh` and setting up the environment via `export.sh`.

Would you accept pull request to update it?

## Problem Description - reliance on the abandoned Python 2 environment

Related to the documentation issues above, the RTOS SDK focuses on the now obsolete Python 2 environment.

The shebang code in the tools also inconsistently calls for python, python2, and python3, further highlighting that the SDK both requires Python 2 and Python 3 to be present:

`
components/partition_table/test_gen_esp32part_host/gen_esp32part_tests.py:#!/usr/bin/env python
components/partition_table/gen_esp32part.py:#!/usr/bin/env python
components/partition_table/parttool.py:#!/usr/bin/env python
components/partition_table/gen_empty_partition.py:#!/usr/bin/env python
components/esptool_py/esptool/ecdsa/ellipticcurve.py:#! /usr/bin/env python
components/esptool_py/esptool/ecdsa/ecdsa.py:#! /usr/bin/env python
components/esptool_py/esptool/ecdsa/numbertheory.py:#! /usr/bin/env python
components/esptool_py/esptool/test/test_imagegen.py:#!/usr/bin/env python
components/esptool_py/esptool/test/test_esptool.py:#!/usr/bin/env python
components/esptool_py/esptool/espsecure.py:#!/usr/bin/env python
components/esptool_py/esptool/flasher_stub/compare_stubs.py:#!/usr/bin/env python
components/esptool_py/esptool/flasher_stub/wrap_stub.py:#!/usr/bin/env python3
components/esptool_py/esptool/flasher_stub/esptool_test_stub.py:#!/usr/bin/env python
components/esptool_py/esptool/esptool.py:#!/usr/bin/env python
components/esptool_py/esptool/espefuse.py:#!/usr/bin/env python
components/mbedtls/mbedtls/tests/scripts/test_generate_test_code.py:#!/usr/bin/env python3
components/mbedtls/mbedtls/tests/scripts/generate_test_code.py:#!/usr/bin/env python3
components/mbedtls/mbedtls/tests/scripts/check-files.py:#!/usr/bin/env python3
components/mbedtls/mbedtls/scripts/abi_check.py:#!/usr/bin/env python3
components/app_update/otatool.py:#!/usr/bin/env python
components/nvs_flash/nvs_partition_generator/nvs_partition_gen.py:#!/usr/bin/env python
docs/gen-version-specific-includes.py:#!/usr/bin/env python
docs/gen-dxd.py:#!/usr/bin/env python
docs/gen-toolchain-links.py:#!/usr/bin/env python
examples/provisioning/legacy/softap_prov/softap_prov_test.py:#!/usr/bin/env python
examples/protocols/http_server/advanced_tests/scripts/test.py:#!/usr/bin/env python
examples/protocols/http_server/simple/http_server_simple_test.py:#!/usr/bin/env python
examples/protocols/http_server/simple/scripts/client.py:#!/usr/bin/env python
examples/protocols/http_server/persistent_sockets/scripts/adder.py:#!/usr/bin/env python
tools/ldgen/test/test_fragments.py:#!/usr/bin/env python
tools/ldgen/test/test_generation.py:#!/usr/bin/env python
tools/ldgen/ldgen.py:#!/usr/bin/env python
tools/gen_esp_err_to_name.py:#!/usr/bin/env python
tools/cmake/convert_to_cmake.py:#!/usr/bin/env python
tools/idf_monitor.py:#!/usr/bin/env python
tools/ci/envsubst.py:#!/usr/bin/env python
tools/check_kconfigs.py:#!/usr/bin/env python
tools/idf.py:#!/usr/bin/env python3
tools/idf_tools.py:#!/usr/bin/env python3
tools/test_check_kconfigs.py:#!/usr/bin/env python
tools/idf_size.py:#!/usr/bin/env python
tools/check_python_dependencies.py:#!/usr/bin/env python3
tools/kconfig_new/test/test_confserver.py:#!/usr/bin/env python
tools/kconfig_new/confserver.py:#!/usr/bin/env python
tools/kconfig_new/confgen.py:#!/usr/bin/env python
tools/kconfig_new/gen_kconfig_doc.py:#!/usr/bin/env python
tools/windows/eclipse_make.py:#!/usr/bin/env python
tools/pack_fw.py:#!/usr/bin/env python2
tools/esp_prov/esp_prov.py:#!/usr/bin/env python
`

However, it seems simply editing the shebang in the following files are enough to run the stack on Python 3.
`
	modified:   tools/check_python_dependencies.py
	modified:   tools/idf.py
	modified:   tools/idf_tools.py
`

NB, [Python 2 has been explicitly made abandoned over a year and a half ago](https://www.python.org/doc/sunset-python-2), and there is no good reason to force people to hack around to try to run the SDK on Python 2. Python 3 is available on all systems by now and the SDK should be ported to Python to ASAP as well.

st1vms, 2022-03-30:
> However, it seems simply editing the shebang in the following files are enough to run the stack on Python 3. `modified: tools/check_python_dependencies.py modified: tools/idf.py modified: tools/idf_tools.py`
> 

I tried this very step, by also installing the library using `install.sh` and `export.sh` it fixed the `python : Command not found` for my python3 environment. :1st_place_medal:

colesnicov, 2025-10-15:
- HP Omen 15
- Ubuntu 25.04
- GNU Make 4.4.1
- SDK8266-RTOS-SDK d412ac601befc4dd024d2d2adcfaa319c7463e36

I've been trying to run the `make menuconfig` command for two days now. I installed `Python2.7` and `pip` manually. I changed the symbolic links so that `python2.7` would be launched when python was called. But I'm stuck on `setuptools` which is apparently outdated.

```
denis:~/$ cp -r $IDF_PATH/examples/get-started/hello_world 
denis:~/$ cd hello_world
```

```
denis:~/hello_world$ . ../ESP8266_RTOS_SDK/export.sh 
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
The following Python requirements are not satisfied:
click>=5.0
pyserial>=3.0
future>=0.15.2
cryptography>=2.1.4,<35
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
To install the missing packages, please run "/home/denis/ESP8266_RTOS_SDK/install.sh"
Alternatively, you can run "/home/denis/.espressif/python_env/rtos3.4_py2.7_env/bin/python -m pip install -r /home/denis/ESP8266_RTOS_SDK/requirements.txt" for resolving the issue.
```

```
denis:~/hello_world$ /home/denis/.espressif/python_env/rtos3.4_py2.7_env/bin/python -m pip install -r /home/denis/ESP8266_RTOS_SDK/requirements.txt
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Requirement already satisfied: setuptools in /home/denis/.espressif/python_env/rtos3.4_py2.7_env/lib/python2.7/site-packages (from -r /home/denis/ESP8266_RTOS_SDK/requirements.txt (line 4)) (44.1.1)
Collecting click>=5.0
  Using cached click-7.1.2-py2.py3-none-any.whl (82 kB)
Collecting pyserial>=3.0
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting future>=0.15.2
  Using cached future-1.0.0.tar.gz (1.2 MB)
Collecting cryptography<35,>=2.1.4
  Using cached cryptography-3.3.2-cp27-cp27m-manylinux2010_x86_64.whl (2.6 MB)
Collecting pyparsing<2.4.0,>=2.0.3
  Using cached pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
Collecting pyelftools>=0.22
  Using cached pyelftools-0.32.tar.gz (15.0 MB)
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /home/denis/.espressif/python_env/rtos3.4_py2.7_env/bin/python /home/denis/.espressif/python_env/rtos3.4_py2.7_env/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-0BKeXA/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools >= 46.4.0'
       cwd: None
  Complete output (3 lines):
  DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
  ERROR: Could not find a version that satisfies the requirement setuptools>=46.4.0 (from versions: 0.6b1, 0.6b2, 0.6b3, 0.6b4, 0.6rc1, 0.6rc2, 0.6rc3, 0.6rc4, 0.6rc5, 0.6rc6, 0.6rc7, 0.6rc8, 0.6rc9, 0.6rc10, 0.6rc11, 0.7.2, 0.7.3, 0.7.4, 0.7.5, 0.7.6, 0.7.7, 0.7.8, 0.8, 0.9, 0.9.1, 0.9.2, 0.9.3, 0.9.4, 0.9.5, 0.9.6, 0.9.7, 0.9.8, 1.0, 1.1, 1.1.1, 1.1.2, 1.1.3, 1.1.4, 1.1.5, 1.1.6, 1.1.7, 1.2, 1.3, 1.3.1, 1.3.2, 1.4, 1.4.1, 1.4.2, 2.0, 2.0.1, 2.0.2, 2.1, 2.1.1, 2.1.2, 2.2, 3.0, 3.0.1, 3.0.2, 3.1, 3.2, 3.3, 3.4, 3.4.1, 3.4.2, 3.4.3, 3.4.4, 3.5, 3.5.1, 3.5.2, 3.6, 3.7, 3.7.1, 3.8, 3.8.1, 4.0, 4.0.1, 5.0, 5.0.1, 5.0.2, 5.1, 5.2, 5.3, 5.4, 5.4.1, 5.4.2, 5.5, 5.5.1, 5.6, 5.7, 5.8, 6.0.1, 6.0.2, 6.1, 7.0, 8.0, 8.0.1, 8.0.2, 8.0.3, 8.0.4, 8.1, 8.2, 8.2.1, 8.3, 9.0, 9.0.1, 9.1, 10.0, 10.0.1, 10.1, 10.2, 10.2.1, 11.0, 11.1, 11.2, 11.3, 11.3.1, 12.0, 12.0.1, 12.0.2, 12.0.3, 12.0.4, 12.0.5, 12.1, 12.2, 12.3, 12.4, 13.0.1, 13.0.2, 14.0, 14.1, 14.1.1, 14.2, 14.3, 14.3.1, 15.0, 15.1, 15.2, 16.0, 17.0, 17.1, 17.1.1, 18.0, 18.0.1, 18.1, 18.2, 18.3, 18.3.1, 18.3.2, 18.4, 18.5, 18.6, 18.6.1, 18.7, 18.7.1, 18.8, 18.8.1, 19.0, 19.1, 19.1.1, 19.2, 19.3, 19.4, 19.4.1, 19.5, 19.6, 19.6.1, 19.6.2, 19.7, 20.0, 20.1, 20.1.1, 20.2.2, 20.3, 20.3.1, 20.4, 20.6.6, 20.6.7, 20.6.8, 20.7.0, 20.8.0, 20.8.1, 20.9.0, 20.10.1, 21.0.0, 21.1.0, 21.2.0, 21.2.1, 21.2.2, 22.0.0, 22.0.1, 22.0.2, 22.0.4, 22.0.5, 23.0.0, 23.1.0, 23.2.0, 23.2.1, 24.0.0, 24.0.1, 24.0.2, 24.0.3, 24.1.0, 24.1.1, 24.2.0, 24.2.1, 24.3.0, 24.3.1, 25.0.0, 25.0.1, 25.0.2, 25.1.0, 25.1.1, 25.1.2, 25.1.3, 25.1.4, 25.1.5, 25.1.6, 25.2.0, 25.3.0, 25.4.0, 26.0.0, 26.1.0, 26.1.1, 27.0.0, 27.1.0, 27.1.2, 27.2.0, 27.3.0, 27.3.1, 28.0.0, 28.1.0, 28.2.0, 28.3.0, 28.4.0, 28.5.0, 28.6.0, 28.6.1, 28.7.0, 28.7.1, 28.8.0, 28.8.1, 29.0.0, 29.0.1, 30.0.0, 30.1.0, 30.2.0, 30.2.1, 30.3.0, 30.4.0, 31.0.0, 31.0.1, 32.0.0, 32.1.0, 32.1.1, 32.1.2, 32.1.3, 32.2.0, 32.3.0, 32.3.1, 33.1.0, 33.1.1, 34.0.0, 34.0.1, 34.0.2, 34.0.3, 34.1.0, 34.1.1, 34.2.0, 34.3.0, 34.3.1, 34.3.2, 34.3.3, 34.4.0, 34.4.1, 35.0.0, 35.0.1, 35.0.2, 36.0.1, 36.1.0, 36.1.1, 36.2.0, 36.2.1, 36.2.2, 36.2.3, 36.2.4, 36.2.5, 36.2.6, 36.2.7, 36.3.0, 36.4.0, 36.5.0, 36.6.0, 36.6.1, 36.7.0, 36.7.1, 36.7.2, 36.8.0, 37.0.0, 38.0.0, 38.1.0, 38.2.0, 38.2.1, 38.2.3, 38.2.4, 38.2.5, 38.3.0, 38.4.0, 38.4.1, 38.5.0, 38.5.1, 3 [... cut ...]
ERROR: Command errored out with exit status 1: /home/denis/.espressif/python_env/rtos3.4_py2.7_env/bin/python /home/denis/.espressif/python_env/rtos3.4_py2.7_env/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-0BKeXA/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'setuptools >= 46.4.0' Check the logs for full command output.

```

**It's probably wrong that I can buy a chip, the ESP8266 is still on the market and there are still a lot of them, but I can't program it.**