espressif/ESP8266_RTOS_SDK #1131: deep sleep current up to 200uA sometimes

rainingsoon, 2021-09-07:
RTOS SDK master version (20210907)

Normally, after esp_deep_sleep(time) called, the ESP8266 goes to deep sleep, the deep sleep current should be about 20 uA, but sometimes the current could be about 200 uA, till next wakeup. After next wakeup, the device goes to deep sleep once more, it could be 20 uA, or it could be 200 uA, no significant clue to show what reason related.

Based on the same hardware, we burned a NORTOS_SDK program on it, deep sleep current was always be 20uA. So, it should be bugs in latest RTOS SDK.

rainingsoon, 2021-09-09:
After some tracing, we found the deep sleep current is related with SPI Flash. Same board with different flash chip, the current is different. W25Q32 is 200uA, MX25Q32 is 2mA. Both QIO and DIO modes are not normal.

rainingsoon, 2021-09-10:
We tested all RTOS_SDK version from v3.0 to v3.4, the deep sleep current was not normal sometimes after deep sleep. For some flash chip (MX25Q) the abnormal current is about 2mA, for some others(e.g. W25Q) the abnormal current is 200uA. Then we tried RTOS_SDK version 2.1, it is all time normal to 20uA for all flashes. We decide to porting our codes from IDF RTOS_SDK to traditional RTOS_SDK v2.1. We hope Espressif task force could fix the bug in time, as soon as possible. Thanks!

I found some clue for the bug, I guess it should be at somewhere in code of deep_sleep preparation part. The firmware should send power down command to SPI flash and check the status.  For some new SPI flash parts, there are new commands to support status checking (1 byte, 2 bytes, 3 bytes, etc.). I did not find such kind of source code in SDK, so, I guess it is located in binary libs. So, currently, we can do nothing without support from Espressif.

MarcinBa, 2022-10-24:
I'm facing the same issue. Even without any peripherals, with minimal configuration most of the time deep sleep consume around 200 uA, but sometimes it goes to 20uA (as expected). Did you find anything more?