espressif/ESP8266_RTOS_SDK #1135: xtensa-lx106-elf-gcc

exrector, 2021-09-27:
Hi. I am facing errors and cannot fix them. xtensa-lx106-elf-gcc can be fixed using ./install.sh but other errors appear. Please help me understand. All software versions are fresh

`MacExrector:ESP8266_RTOS_SDK Exrector$ export IDF_PATH=~/esp/ESP8266_RTOS_SDK
MacExrector:ESP8266_RTOS_SDK Exrector$ cd ~/esp/ESP8266_RTOS_SDK/examples/get-started/hello_world
MacExrector:hello_world Exrector$ make menuconfig
make: xtensa-lx106-elf-gcc: Command not found
expr: syntax error
/bin/sh: xtensa-lx106-elf-gcc: command not found
/bin/sh: xtensa-lx106-elf-gcc: command not found
Toolchain path: 
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
Python requirements from /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
MENUCONFIG
configuration written to /Users/exrector/esp/ESP8266_RTOS_SDK/examples/get-started/hello_world/sdkconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

MacExrector:hello_world Exrector$ make all
make: xtensa-lx106-elf-gcc: Command not found
expr: syntax error
/bin/sh: xtensa-lx106-elf-gcc: command not found
/bin/sh: xtensa-lx106-elf-gcc: command not found
Toolchain path: 
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
Python requirements from /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
App "hello-world" version: v3.4-49-g0f200b46
make: xtensa-lx106-elf-gcc: Command not found
expr: syntax error
Building partitions from /Users/exrector/esp/ESP8266_RTOS_SDK/components/partition_table/partitions_singleapp.csv...
Python requirements from /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
make[1]: xtensa-lx106-elf-gcc: Command not found
expr: syntax error
/bin/sh: xtensa-lx106-elf-gcc: command not found
/bin/sh: xtensa-lx106-elf-gcc: command not found
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
make[1]: xtensa-lx106-elf-gcc: Command not found
expr: syntax error
CC build/bootloader/main/bootloader_start.o
make[2]: xtensa-lx106-elf-gcc: No such file or directory
make[2]: *** [bootloader_start.o] Error 1
make[1]: *** [component-main-build] Error 2
make: *** [/Users/exrector/esp/ESP8266_RTOS_SDK/examples/get-started/hello_world/build/bootloader/bootloader.bin] Error 2`

**var 2**

`MacExrector:ESP8266_RTOS_SDK Exrector$ ./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-lx106-elf
Skipping xtensa-lx106-elf@esp-2020r3-49-gd5524c1-8.4.0 (already installed)
Installing Python environment and packages
Installing Python packages from /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Requirement already satisfied: setuptools in /Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages (from -r /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt (line 4)) (41.2.0)
Requirement already satisfied: click>=5.0 in /Users/exrector/Library/Python/2.7/lib/python/site-packages (from -r /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt (line 8)) (7.1.2)
Requirement already satisfied: pyserial>=3.0 in /Users/exrector/Library/Python/2.7/lib/python/site-packages (from -r /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in /Users/exrector/Library/Python/2.7/lib/python/site-packages (from -r /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /Users/exrector/Library/Python/2.7/lib/python/site-packages (from -r /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt (line 11)) (3.1)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /Users/exrector/Library/Python/2.7/lib/python/site-packages (from -r /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /Users/exrector/Library/Python/2.7/lib/python/site-packages (from -r /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt (line 13)) (0.26)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /Users/exrector/Library/Python/2.7/lib/python/site-packages (from cryptography>=2.1.4->-r /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt (line 11)) (1.14.2)
Requirement already satisfied: ipaddress; python_version < "3" in /Users/exrector/Library/Python/2.7/lib/python/site-packages (from cryptography>=2.1.4->-r /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt (line 11)) (1.0.23)
Requirement already satisfied: six>=1.4.1 in /Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt (line 11)) (1.15.0)
Requirement already satisfied: enum34; python_version < "3" in /Users/exrector/Library/Python/2.7/lib/python/site-packages (from cryptography>=2.1.4->-r /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt (line 11)) (1.1.10)
Requirement already satisfied: pycparser in /Users/exrector/Library/Python/2.7/lib/python/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /Users/exrector/esp/ESP8266_RTOS_SDK/requirements.txt (line 11)) (2.20)
All done! You can now run:

  . ./export.sh

MacExrector:ESP8266_RTOS_SDK Exrector$   . ./export.sh
[... 37 lines cut ...]
CC build/bootloader/spi_flash/src/spi_flash.o
CC build/bootloader/spi_flash/src/spi_flash_raw.o
CC build/bootloader/spi_flash/port/port.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/esp8266/source/ets_printf.o
CC build/bootloader/esp8266/source/esp_fast_boot.o
AR build/bootloader/esp8266/libesp8266.a
LD build/bootloader/bootloader.elf
Traceback (most recent call last):
  File "/Users/exrector/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py", line 35, in <module>
    import serial.tools.list_ports as list_ports
  File "/Users/exrector/Library/Python/2.7/lib/python/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Users/exrector/Library/Python/2.7/lib/python/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Users/exrector/Library/Python/2.7/lib/python/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
make[1]: *** [/Users/exrector/esp/ESP8266_RTOS_SDK/examples/get-started/hello_world/build/bootloader/bootloader.bin] Error 1
make: *** [/Users/exrector/esp/ESP8266_RTOS_SDK/examples/get-started/hello_world/build/bootloader/bootloader.bin] Error 2
MacExrector:hello_world Exrector$ `

**finita la commedia**

exrector, 2021-09-28:
тут я разобрался

closed 2021-09-28 by exrector

moefear85, 2021-12-10:
looks like you have python2 setup in your system rather than python3. That is most likely the error, and I don't think esp8266_rtos_sdk works with python2. You'll have to upgrade python to python3 first, then run install.sh and export.sh.