espressif/ESP8266_RTOS_SDK #1147: Guru Meditation Error: Core  0 panic'ed (unknown). Exception was unhandled.

zenz, 2021-10-31:
## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 0f200b46
- Development Env:      [Arduino IDE|Make|Eclipse|other] Make
- Operating System:     [Windows|Ubuntu|MacOS] MacOS Big Sur 11.6
- Power Supply:         [USB|external 5V|external 3.3V|Battery] external 3.3V

## Problem Description

make monitor will raise a crash after awhile

### Expected Behavior

continous to work

### Actual Behavior
Task watchdog got triggered.

Guru Meditation Error: Core  0 panic'ed (unknown). Exception was unhandled.
Core 0 register dump:
PC      : 0x4010041a  PS      : 0x00000030  A0      : 0x4021267c  A1      : 0x3ffea0f0

### Steps to repropduce

1. make menuconfig
2. make
3. make flash
4. make monitor

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"

#define GPIO_LED 5
#define GPIO_LED_PIN_SEL (1ULL << GPIO_LED)

static void init_hw(void)
{
    gpio_config_t io_conf;
    io_conf.mode = GPIO_MODE_OUTPUT;
    io_conf.pin_bit_mask = GPIO_LED_PIN_SEL;
    io_conf.intr_type = GPIO_INTR_DISABLE;
    io_conf.pull_down_en = 0;
    io_conf.pull_up_en = 0;
    gpio_config(&io_conf);
}

void app_main()
{
    TickType_t next = 0;
    bool led_state = false;

    printf("Hello world!\n");
    init_hw();
    while (1)
    {
        if (xTaskGetTickCount() >= next)
        {
            next = xTaskGetTickCount() + 500 / portTICK_PERIOD_MS;
            led_state = !led_state;
            gpio_set_level(GPIO_LED, led_state);
        }
    }
}

```
## Debug Logs

```
 make monitor
Toolchain path: /Users/simon/esp-idf/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: esp-2020r3-49-gd5524c1
Compiler version: 8.4.0
Python requirements from /Users/simon/esp-idf/ESP8266_RTOS_SDK/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/cu.usbserial-0001 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7032, room 16 
tail 8
chksum 0x68
load 0x3ffe8408, len 24, room 0 
tail 8
chksum 0xe1
load 0x3ffe8420, len 3324, room 0 
tail 12
chksum 0xc7
csum 0xc7
I (42) boot: ESP-IDF v3.4-49-g0f200b46 2nd stage bootloader
I (43) boot: compile time 23:08:33
I (43) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 2MB
I (61) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (78) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (90) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (101) boot:  2 factory          factory app      00 00 00010000 000f0000
I (113) boot: End of partition table
I (119) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x16058 ( 90200) map
0x40210010: _stext at ??:?

I (170) esp_image: segment 1: paddr=0x00026070 vaddr=0x40226068 size=0x05958 ( 22872) map
I (180) esp_image: segment 2: paddr=0x0002b9d0 vaddr=0x3ffe8000 size=0x003dc (   988) load
I (181) esp_image: segment 3: paddr=0x0002bdb4 vaddr=0x40100000 size=0x00080 (   128) load
I (195) esp_image: segment 4: paddr=0x0002be3c vaddr=0x40100080 size=0x04558 ( 17752) load
I (216) boot: Loaded app from partition at offset 0x10000
Hello world!
I (237) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Task watchdog got triggered.

Guru Meditation Error: Core  0 panic'ed (unknown). Exception was unhandled.
Core 0 register dump:
PC      : 0x4010041a  PS      : 0x00000030  A0      : 0x4021267c  A1      : 0x3ffea0f0  
0x4010041a: vPortExitCritical at /Users/simon/esp-idf/ESP8266_RTOS_SDK/components/freertos/port/esp8266/port.c:263

0x4021267c: xTaskGetTickCount at /Users/simon/esp-idf/ESP8266_RTOS_SDK/components/freertos/freertos/tasks.c:2252

A2      : 0x00000020  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x60000300  
A6      : 0x00000000  A7      : 0x4021b7dc  A8      : 0x00000001  A9      : 0x00000090  
0x4021b7dc: uart_write at /Users/simon/esp-idf/ESP8266_RTOS_SDK/components/vfs/vfs_uart.c:210

A10     : 0x4022953e  A11     : 0x00000020  A12     : 0x00000a7d  A13     : 0x00000a8e  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000a8e  

Backtrace: 0x4010041a:0x3ffea0f0 0x4021267c:0x3ffea100 0x40213347:0x3ffea110 0x40210a0b:0x3ffea120 
0x4010041a: vPortExitCritical at /Users/simon/esp-idf/ESP8266_RTOS_SDK/components/freertos/port/esp8266/port.c:263

0x4021267c: xTaskGetTickCount at /Users/simon/esp-idf/ESP8266_RTOS_SDK/components/freertos/freertos/tasks.c:2252

0x40213347: app_main at /Users/simon/Programs/airtub_partner_rtos/main/airtub_partner_rtos.c:37

0x40210a0b: user_init_entry at /Users/simon/esp-idf/ESP8266_RTOS_SDK/components/esp8266/source/startup.c:98
```

KaeLL, 2021-10-31:
> Task watchdog got triggered.

Either disable it, or have the task yield with vtaskdelay

zenz, 2021-10-31:
> > Task watchdog got triggered.
> 
> Either disable it, or manually yield to other tasks with vtaskdelay

Thank you @KaeLL , it seems vTaskDelay(1) fix the issue. I'm very new to FreeRTOS. Thank you again.

closed 2021-10-31 by zenz