espressif/ESP8266_RTOS_SDK #1166: ESP8266 RTOS SDK Support Period Policy

Eavo, 2022-01-06:
ESP8266 RTOS SDK 不会再发布新的主版本，所以不再添加新的功能，如非必要也不会更新 components 里面的组件。根据 [ESP8266 RTOS SDK 的支持政策](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/SUPPORT_POLICY_CN.md)，请相关用户（尤其是新用户）使用还在维护期的分支（release/v3.4）和版本（v3.4）。

ESP8266 RTOS SDK will not release new major version, so it will not add new feature or update components unless necessary. According to the [ESP8266 RTOS SDK Support Period Policy](https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/SUPPORT_POLICY_EN.md), please use the branch (release/v3.4) or the tag (v3.4), which are still in maintenance, especially for new users.

ccrause, 2022-03-18:
Does this mean that ESP8266 support will be added to esp-idf?

nicklasfrahm, 2022-03-30:
I would be more than happy to help merging this SDK into the ESP IDF, if that meant that the ESP IDF can be used for the ESP8266 and ESP32 family.

luzidchris, 2022-12-10:
> I would be more than happy to help merging this SDK into the ESP IDF, if that meant that the ESP IDF can be used for the ESP8266 and ESP32 family.

I also hope that'll some day be possible. In the past, the excuse often was that esp-idf wasn't capable of handling multiple architectures but since support of the esp32 variants based on RISC-V was added to esp-idf, that's not really valid anymore.

I understand from Espressif's perspective that they'd like to solely focus on ESP32 but there are so many ESP8266 out there... and IMHO ESP8266 is still a capable device used by many many people, so there should be enough support from the community.

I assume that at least "bootloader", "esp8266", "freertos" and "wpa_supplicant" are target-specific and modified versions of these components need to be added to esp-idf to add support of ESP8266 to it. Then there's the toolchain-esp8266.cmake. Anything else? Why haven't there been any attempts to use esp-idf for ESP8266 so far? Although the build system is pretty similar, it's obviously not enough to simply add the mentioned components and cmake file(s) of ESP8266_RTOS_SDK to esp-idf. What exactly is the blocker from a technical stand-point?