espressif/ESP8266_RTOS_SDK #1169: i2s transmit

ttennebb, 2022-01-19:
https://github.com/espressif/ESP8266_RTOS_SDK/blob/83517ba1f5e26b9413f2ef6d2503dd73662b1272/components/esp8266/driver/i2s.c#L750

When I tried the example code for i2s, the esp8266 would hang. After much gnashing of teeth, I finally found the culprit.
In the subroutine, esp_err_t i2s_write(), the line : 
            if (xQueueReceive(p_i2s_obj[i2s_num]->tx->queue, &p_i2s_obj[i2s_num]->tx->curr_ptr, ticks_to_wait) == pdFALSE) {
should be : 
            if (xQueueSend(p_i2s_obj[i2s_num]->tx->queue, &p_i2s_obj[i2s_num]->tx->curr_ptr, ticks_to_wait) == pdFALSE) {

ttennebb, 2022-01-19:
I found it because I found no errors on the initial runs, but no data/clock coming out any gpio.
First blush, the original line with xQueueReceive would fail (of course) then break out of the while loop, give back the semaphore, returning ESP_OK. No error being logged or handed back.

        if (p_i2s_obj[i2s_num]->tx->rw_pos == p_i2s_obj[i2s_num]->tx->buf_size || p_i2s_obj[i2s_num]->tx->curr_ptr == NULL) {        
            if (xQueueReceive(p_i2s_obj[i2s_num]->tx->queue, &p_i2s_obj[i2s_num]->tx->curr_ptr, ticks_to_wait) == pdFALSE) {
                break;
            }
            p_i2s_obj[i2s_num]->tx->rw_pos = 0;
        }

This will throw an error in a log or on screen when monitoring :

        if (p_i2s_obj[i2s_num]->tx->rw_pos == p_i2s_obj[i2s_num]->tx->buf_size || p_i2s_obj[i2s_num]->tx->curr_ptr == NULL) {        
            if (xQueueReceive(p_i2s_obj[i2s_num]->tx->queue, &p_i2s_obj[i2s_num]->tx->curr_ptr, ticks_to_wait) == pdFALSE) {
                ESP_LOGE(I2S_TAG, "Time out waiting for ticks_to_wait : %d", ticks_to_wait);
                break;
            }
            p_i2s_obj[i2s_num]->tx->rw_pos = 0;
        }

That would at least send out some notification as to where to look.
Or better still :

        I2S_CHECK((xQueueReceive(p_i2s_obj[i2s_num]->tx->queue, &p_i2s_obj[i2s_num]->tx->curr_ptr, ticks_to_wait) == pdTRUE), "Time out waiting for ticks_to_wait", ESP_ERR_TIMEOUT);

Hope this helps. Thanks for all your efforts!

tieutuanbao, 2022-04-06:
Too long time for this issue ....

tieutuanbao, 2022-04-06:
I tried replace I2S.c master by I2S Branch V3.2, it's working!