espressif/ESP8266_RTOS_SDK #1174: Incomplete toollchain setup instructions.

jimlynnjulian, 2022-03-26:
I attempted to install the toolchain according to the instructions. At the bottom of the first page was supposed to be a link to the continuation of the process. Instead, the link is to the same [page on which the link occurs, i.e. self-referencing.
Where is the continuation? I'd like to finnish the toolchain setup. I downloaded the Extensa toolset and installed in a parallel directory to RTOS_SDK. Both are in the home/Jim/esp subdirectory of MSYS2.

bobmc-rmm, 2022-04-02:
I am trying to setup ESP8266_RTOS_SDK rel 3.4.  I failed for a couple or reasons.

-1  The instructions on the web page are incomplete. Later I found better instructions /tools/docs/en. Well ok.

-2  Using Xubuntu Linux, the apt-get causes a switch from python3.9 to python2 which messes with the Linux installation.  There is no need to install python on Ubuntu since Python2 and Python3 are available. Legacy scripts need to name Python2 explicitly since Python3 is the new Python by default.

-3  I appreciate having the Espressif SDKs ... but they are so tedious.

sudo apt-get install gcc git wget make libncurses-dev flex bison gperf python python-serial
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'python-is-python2' instead of 'python'
E: Unable to locate package python-serial

LaXiS96, 2022-11-10:
Follow the setup instructions detailed here: https://espressif-docs.readthedocs-hosted.com/projects/esp8266-rtos-sdk/en/latest/get-started/index.html

One thing not explained in the docs is that you need to extract the contents of the `xtensa-lx106-elf` folder from `xtensa-lx106-elf-gcc8_4_0-esp-2020r3-win32.zip` into `C:\msys32\mingw32\`.

I recommend not to update this MSYS2 installation and only use it as is.