espressif/ESP8266_RTOS_SDK #1196: ¨MQTT_CLIENT: Error transport connect¨ when trying to run example mqtt/tcp

KelvinyHenrique, 2022-09-24:
## Environment

- Development Kit:
https://pt.aliexpress.com/item/1005004165920862.html
My board is ESP-01s (ESP8266EX)

- IDF version: 
7085cc76
- Development Env:      [VSCODE IDE|Make]
- Operating System:     [Ubuntu]
- Power Supply:         [USB]

## Problem Description
MQTT protocol over tcp example does not work

I built the original example, the changes I made were in the menuconfig, I set my ssid and password to connect the wifi, I also changed the Log output to debug to try to identify the error.

I always get the error MQTT_CLIENT: Error transport connect in an infinite loop.

### Code to reproduce this issue

Example: https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/protocols/mqtt

## Debug Logs

```
-- idf_monitor on /dev/ttyUSB0 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (1931) wifi:state: 0 -> 2 (b0)
I (1937) wifi:state: 2 -> 3 (0)
I (1941) wifi:state: 3 -> 5 (10)
I (2128) wifi:connected with SIGNET_ROOT, aid = 5, channel 1, HT20, bssid = c0:9f:e1:b6:e8:00
D (2129) system_event: SYSTEM_EVENT_STA_CONNECTED, ssid:SIGNET_ROOT, ssid_len:11, bssid:c0:9f:e1:b6:e8:00, channel:1, authmode:3
D (2145) event: running post WIFI_EVENT:4 with handler 0x4024a9d4 on loop 0x40106f50
0x4024a9d4: handle_sta_connected at /home/kelviny/esp/ESP8266_RTOS_SDK/components/tcpip_adapter/event_handlers.c:86

D (2154) tcpip_adapter: dhcp client init ip/mask/gw to all-0
D (2162) tcpip_adapter: if0 start ip lost tmr: enter
D (2170) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x4010721c interval=120 ip=0
D (2183) tcpip_adapter: enable wifi ps
D (2194) tcpip_adapter: dhcp client start successfully
D (4966) tcpip_adapter: if0 dhcpc cb
D (4968) tcpip_adapter: if0 ip changed=1
D (4971) system_event: SYSTEM_EVENT_STA_GOT_IP, ip:192.168.50.6, mask:255.255.255.0, gw:192.168.50.254
D (4976) event: running post IP_EVENT:0 with handler 0x4024a8b8 on loop 0x40106f50
0x4024a8b8: handle_sta_got_ip at /home/kelviny/esp/ESP8266_RTOS_SDK/components/tcpip_adapter/event_handlers.c:44

I (4987) tcpip_adapter: sta ip: 192.168.50.6, mask: 255.255.255.0, gw: 192.168.50.254
D (5000) event: running post IP_EVENT:0 with handler 0x4022b97c on loop 0x40106f50
0x4022b97c: on_got_ip at /home/kelviny/esp/ESP8266_RTOS_SDK/examples/common_components/protocol_examples_common/connect.c:73

D (5011) tcpip_adapter: disable wifi ps
I (5017) example_connect: Connected to SIGNET_ROOT
I (5026) example_connect: IPv4 address: 192.168.50.6
I (5035) system_api: Base MAC address is not set, read default base MAC address from EFUSE
D (5048) nvs: nvs_open_from_partition backup_mac 0
D (5055) nvs: nvs_get_str_or_blob backup_mac_data
D (5062) nvs: nvs_close 7
D (5065) MQTT_CLIENT: MQTT client_id=ESP32_a57E2F
D (5072) event: created event loop 0x40107a14
D (5078) MQTT_CLIENT: Core selection disabled
D (5085) event: running post MQTT_EVENTS:7 with handler 0x402226d4 on loop 0x40107a14
0x402226d4: mqtt_event_handler at /home/kelviny/esp/tcp/main/app_main.c:86

D (5096) MQTT_EXAMPLE: Event dispatched from event loop base=MQTT_EVENTS, event_id=7
I (5108) MQTT_EXAMPLE: Other event id:7
E (5615) MQTT_CLIENT: Error transport connect

```

[... 511 lines cut ...]
CONFIG_TCP_MSS=1440
CONFIG_TCP_MSL=60000
CONFIG_TCP_SND_BUF_DEFAULT=2880
CONFIG_TCP_WND_DEFAULT=5760
CONFIG_TCP_RECVMBOX_SIZE=6
CONFIG_TCP_QUEUE_OOSEQ=y
# CONFIG_ESP_TCP_KEEP_CONNECTION_WHEN_IP_CHANGES is not set
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=2560
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
# End of deprecated options

```

fadushin, 2022-12-27:
I am having the same problem.  The ESP8266 is issued an IP address from the router (PfSense), but no outbound connections are working.  I am getting a timeout in DNS resolution (`gethostbyname`) if I use an address, or in `connect`, if I am using a raw IP address.

The ESP8266 is not responding to ICMP pings.

fadushin, 2022-12-27:
I was able to solve this by fixing the default MQTT host to `mqtt.eclipseprojects.io`